mantidproject/mantid #34082: Instrument View Pick Tab - pixel selection only on mouse click

DannyHindson, 2022-06-14:
This relates to the pixel selection tool on the pick tab of the instrument viewer.

This used to select the pixel that the mouse was hovering over in 6.3 but now the user has to click on a pixel.

Bit hard to illustrate this in a screenshot but here's the pick tab with the workspace I was using during this test

[image]

As well as being a change in behaviour it means that when the user switches from "Pixel selection" to the "Whole instrument" tool the detector details don't clear in the Selection text box. This used to happen because the move of the mouse over to the "whole instrument" button used to deselect the detector

martyngigg, 2022-06-14:
@MathieuTillet @darominski Was there an intention to change the default behaviour of the instrument view pick tab?

darominski, 2022-06-14:
@MathieuTillet is currently looking into this.

MathieuTillet, 2022-06-14:
That should fix it. Thanks for spotting this !

martyngigg, 2022-06-14:
Thanks for the quick fix!

PR #34087 (2022-06-14): Pick info on hovering in the instrument view
**Description of work.**
This PR is a fix for a previous feature that was broken by accident when introducing the tool tip. When touching a component in the pick tab, the left side info is now once again immediately displayed.

**To test:**
- Open a file in the Instrument View
- Go to the pick tab
- Hover over the instrument, and check the values on the left are updated without a click being needed
- In the render tab, check the tool tip option still works when checked, and does not interfere with the pick tab display

Fixes #34082 . 

*This does no [... cut]