mantidproject/mantid #34083: Sliceviewer grids cannot be enabled with lineplots

DanielMurphy22, 2022-06-14:
Found in Smoke testing: https://github.com/mantidproject/mantid/issues/34075#issuecomment-1154952406
Regression
**Describe the bug**
On Sliceviewer grid lines cannot be enabled while lineplots are

**To Reproduce**
Open Sliceviewer on any data, 
enable lineplots in the toolbar options
try to enable gridlines

**Expected behavior**
Gridlines and lineplots work and do not affect eachother

**Platform/Version (please complete the following information):**
 - OS: all
 - Mantid Version 6.4 release candidate

martyngigg, 2022-06-14:
@darominski Would there be any chance you or someone at ILL would be free to look at this?

MialLewis, 2022-06-14:
Caused by this PR https://github.com/mantidproject/mantid/pull/33363

mantid-builder, 2022-06-15:
Fixed by #34092

PR #34092 (2022-06-15): Fix Sliceviewer grids cannot be enabled with lineplots
**Description of work.**
As part of a previous PR a regression was introduced that prevented grid lines from showing when line plots were active on SliceViewer.

This was caused by the image axes x and y axis being set `visible = False` entirely. For grid lines to show, the relevant tick markers must be active.

This PR individually hides axis and marker labels, leaving the ticks visible (though in reality obscured by the line plots as intended). 

**To test:**

Follow the test instructions on the following Issue and PR:
https://github.com [... cut]
closed 2022-06-15 by mantid-builder