mantidproject/mantid #34100: Sliceviewer not closed when MDEventWorkspace deleted

gemmaguest, 2022-06-15:
**Describe the bug**
Raising for info. Not sure how high priority this is.

Sliceviewer is not closed when you delete the MDEventWorkspace (and its dynamically rebinned workspace) that it was opened on.

**To Reproduce**
- Create an MDEventWorkspace e.g. `md_4D = CreateMDWorkspace(Dimensions=4, Extents=[-2,2,-1,1,-1.5,1.5,-0.25,0.25], Names="H,K,L,E", Frames='HKL,HKL,HKL,General Frame',Units='r.l.u.,r.l.u.,r.l.u.,meV')`
- Open sliceviewer. This creates `md_4D_svrebinned`.
- Select both workspaces and delete them.
- Sliceviewer does not close. If you e.g. zoom in on sliceviewer, `md_4D_svrebinned` reappears.

**Expected behavior**
Sliceviewer should close when the workspace it was created on is deleted.

**Platform/Version (please complete the following information):**
 - OS: All
 - Mantid Version 6.3.0, 6.4.0

**Additional context**
If you select Clear to delete the workspaces instead of manually selecting and deleting them, then it works.

RichardWaiteSTFC, 2023-02-08:
I can't reproduce this exactly on IDAaaS using mantid nightly from 1 Feb (i.e sliceviewer does close when I delete both workspaces - because I'm deleting one of the workspaces it is open on) - however, if you open `md_4D_svrebinned` and delete `md_4D` the sliceviewer should close and doesn't (which is a well known bug, I don;t know if there is already an issue for it?)

RichardWaiteSTFC, 2023-02-08:
Yep there is an issue for the bug I mention above #33434

sf1919, 2023-02-08:
I've been unable to recreate the issue as outlined above on Windows either. As the other issue is detailed in #33434 I'm going to close this one.