mantidproject/mantid #34103: wstring_convert() deprecation warning

DanielMurphy22, 2022-06-15:
Found when tackling issue: https://github.com/mantidproject/mantid/issues/33801

**Describe the bug**
As mentioned [here](https://github.com/mantidproject/mantid/issues/33801#issuecomment-1150089236), there is a warning when building with `-DUSE_PRECOMPILED_HEADERS=OFF`.

<details>
    <summary>UnitLabel.cppL28: wstring_convert() deprecation warning</summary>

    C:\Users\mlc47243\Documents\mantid-development\mantid-conda\mantid\Framework\Kernel\src\UnitLabel.cpp(28,29): warning C4996: 'std::codecvt_utf8_utf16<wchar_t,1114111,(std::codecvt_mode)0>': warning STL4017: std::wbuffer_convert, std::wstring_convert, and the <codecvt> header (containing std::codecvt_mode, std::codecvt_utf8, std::codecvt_utf16, and std::codecvt_utf8_utf16) are deprecated in C++17. (The std::codecvt class template is NOT deprecated.) The C++ Standard doesn't provide equivalent non-deprecated functionality; consider using MultiByteToWideChar() and WideCharToMultiByte() from <Windows.h> instead. You can define _SILENCE_CXX17_CODECVT_HEADER_DEPRECATION_WARNING or _SILENCE_ALL_CXX17_DEPRECATION_WARNINGS to acknowledge that you have received this warning. [C:\Users\mlc47243\Documents\mantid-development\mantid-conda\build\Framework\Kernel\Kernel.vcxproj]
    C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.27.29110\include\codecvt(428,5): message : see declaration of 'std::codecvt_utf8_utf16' (compiling source file C:\Users\mlc47243\Documents\mantid-development\mantid-conda\mantid\Framework\Kernel\src\UnitLabel.cpp) [C:\Users\mlc47243\Documents\mantid-development\mantid-conda\build\Framework\Kernel\Kernel.vcxproj]
    C:\Users\mlc47243\Documents\mantid-development\mantid-conda\mantid\Framework\Kernel\src\UnitLabel.cpp(28,8): warning C4996: 'std::wstring_convert<std::codecvt_utf8_utf16<wchar_t,1114111,(std::codecvt_mode)0>,wchar_t,std::allocator<wchar_t>,std::allocator<char>>': warning STL4017: std::wbuffer_convert, std::wstring_convert, and the <codecvt> header (containing std::codecvt_mode, std::codecvt_utf8, std::codecvt_utf16, and std::codecvt_utf8_utf16) are deprecated in C++17. (The std::codecvt class template is NOT deprecated.) The C++ Standard doesn't provide equivalent non-deprecated functionality; consider using MultiByteToWideChar() and WideCharToMultiByte() from <Windows.h> instead. You can define _SILENCE_CXX17_CODECVT_HEADER_DEPRECATION_WARNING or _SILENCE_ALL_CXX17_DEPRECATION_WARNINGS to acknowledge that you have received this warning. [C:\Users\mlc47243\Documents\mantid-development\mantid-conda\build\Framework\Kernel\Kernel.vcxproj]
    C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.27.29110\include\xlocbuf(301,55): message : see declaration of 'std::wstring_convert' (compiling source file C:\Users\mlc47243\Documents\mantid-development\mantid-conda\mantid\Framework\Kernel\src\UnitLabel.cpp) [C:\Users\mlc47243\Documents\mantid-development\mantid-conda\build\Framework\Kernel\Kernel.vcxproj]
    C:\Users\mlc47243\Documents\mantid-development\mantid-conda\mantid\Framework\Kernel\src\UnitLabel.cpp(28,58): warning C4996: 'std::wstring_convert<std::codecvt_utf8_utf16<wchar_t,1114111,(std::codecvt_mode)0>,wchar_t,std::allocator<wchar_t>,std::allocator<char>>::wstring_convert': warning STL4017: std::wbuffer_convert, std::wstring_convert, and the <codecvt> header (containing std::codecvt_mode, std::codecvt_utf8, std::codecvt_utf16, and std::codecvt_utf8_utf16) are deprecated in C++17. (The std::codecvt class template is NOT deprecated.) The C++ Standard doesn't provide equivalent non-deprecated functionality; consider using MultiByteToWideChar() and WideCharToMultiByte() from <Windows.h> instead. You can define _SILENCE_CXX17_CODECVT_HEADER_DEPRECATION_WARNING or _SILENCE_ALL_CXX17_DEPRECATION_WARNINGS to acknowledge that you have received this warning. [C:\Users\mlc47243\Documents\mantid-development\mantid-conda\build\Framework\Kernel\Kernel.vcxproj]
    C:\Users\mlc47243\Documents\mantid-development\mantid-conda\mantid\Framework\Kernel\src\UnitLabel.cpp(29,21): warning C4996: 'std::wstring_convert<std::codecvt_utf8_utf16<wchar_t,1114111,(std::codecvt_mode)0>,wchar_t,std::allocator<wchar_t>,std::allocator<char>>::from_bytes': warning STL4017: std::wbuffer_convert, std::wstring_convert, and the <codecvt> header (containing std::codecvt_mode, std::codecvt_utf8, std::codecvt_utf16, and std::codecvt_utf8_utf16) are deprecated in C++17. (The std::codecvt class template is NOT deprecated.) The C++ Standard doesn't provide equivalent non-deprecated functionality; consider using MultiByteToWideChar() and WideCharToMultiByte() from <Windows.h> instead. You can define _SILENCE_CXX17_CODECVT_HEADER_DEPRECATION_WARNING or _SILENCE_ALL_CXX17_DEPRECATION_WARNINGS to acknowledge that you have received this warning. [C:\Users\mlc47243\Documents\mantid-development\mantid-conda\build\Framework\Kernel\Kernel.vcxproj]  

</details>

This flag isn't too important but the warning may be silenced somewhere else in the codebase. More thought is needed as to whether we need this wstring_convert and sothis deserves a separate issue

**To Reproduce**
run ``cmake -DUSE_PRECOMPILED_HEADERS=OFF ..``

**Additional context**
This funciton is deprecated in C++17 but still exists (deprecated) in C++20 so we could make no code change and silence the warning
We should review the code that is currrently using it to check that it is even needed in the first place.

thomashampson, 2022-08-17:
It was added here:
https://github.com/mantidproject/mantid/pull/31036

DanielMurphy22, 2023-02-01:
Interesting, shall we add a comment to the code with this information and mention the current data when we found this info. Feel free to even put the link to this issue in the code comment. I'd say we can close this issue when we merged the code comment. Can we suppress this warning?

robertapplin, 2023-02-01:
> Interesting, shall we add a comment to the code with this information and mention the current data when we found this info. Feel free to even put the link to this issue in the code comment. I'd say we can close this issue when we merged the code comment. Can we suppress this warning?

We might be able to suppress the warning by defining `_SILENCE_CXX17_CODECVT_HEADER_DEPRECATION_WARNING` which seems specific enough to codecvt for us to use it. And I can leave a comment next to where I define this flag

PR #35126 (2023-02-08): Windows compilation warnings fix
**Description of work.**
This PR fixes two compilation warnings seen on windows. One when building with Ninja, and the other when building with VS.

It also suppresses a deprecation warning seen when building with Ninja on windows. See this issue for more info #34103

**To test:**
1. Build Mantid using the Ninja generator:
```
cmake --preset=win-ninja -B=../buildninja
cd ../buildninja && ninja
```
There should be no warnings

2. Build Mantid using the VS generator:
```
cmake --preset=win-msvc -B=../buildvs
cd ../buildvs && cmake --build .
```
There should be no [... cut]
closed 2023-02-08 by martyngigg