mantidproject/mantid #34125: Sliceviewer doesn't update when QxQz workspace changed

gemmaguest, 2022-06-20:
**Original reporter:** Max, Becky at ISIS

**Describe the bug**
When doing a conversion to Q with ConvertToReflectometryQ, if you open sliceviewer on the result then re-run the algorithm with different settings, sliceviewer does not update.

**To Reproduce**
This example uses a file from the ISIS data archive:

```
Load(Filename='INTER00053105.nxs', OutputWorkspace='INTER00053105')
ConvertUnits(InputWorkspace='INTER00053105', OutputWorkspace='INTER00053105_lam', Target='Wavelength', AlignBins=True)
CropWorkspace(InputWorkspace='INTER00053105_lam', OutputWorkspace='INTER00053105_lam', XMin=1.5, XMax=15, StartWorkspaceIndex=4, EndWorkspaceIndex=70)
ConvertSpectrumAxis(InputWorkspace='INTER00053105_lam', OutputWorkspace='INTER00053105_thlam', Target='signed_theta')
ConvertToReflectometryQ(InputWorkspace='INTER00053105_thlam', Method='NormalisedPolygon', \
                        OverrideIncidentTheta=True, IncidentTheta=0.25, Extents='-0.0001,0.0001,0.01,0.08', \
                        OutputAsMDWorkspace=False, OutputWorkspace='INTER00053105_QxQz', OutputVertexes='INTER00053105_vertexes')

''' OPEN SLICEVIEWER ON INTER00053105_QxQz then run the following
        - sliceviewer does not update
        - open a new sliceviewer on the same workspace and see that the plot is different '''
ConvertToReflectometryQ(InputWorkspace='INTER00053105_thlam', Method='NormalisedPolygon', \
                        OverrideIncidentTheta=True, IncidentTheta=0.5, Extents='-0.0001,0.0001,0.01,0.08', \
                        OutputAsMDWorkspace=False, OutputWorkspace='INTER00053105_QxQz', OutputVertexes='INTER00053105_vertexes')                     
```

**Expected behavior**
Sliceviewer should update when the workspace is changed to show the new data

**Screenshots**
[image]

**Platform/Version (please complete the following information):**
 - OS: All
 - OS Version:
 - Mantid Version 6.3, 6.4

gemmaguest, 2022-06-21:
Breaking in sliceviewer presenter's `replace_handle` it seems there's something weird going on with the notification. When the bug exhibits, I get two notifications about the `vertexes` output workspace instead of one about the vertexes and one for `QxQz`. It seems there's maybe a race condition in the observer and it's notifying about the wrong output workspace. Unfortunately if I break on the C++ side then the bug does not exhibit so it's difficult to work out what's going on.

martyngigg, 2022-09-08:
> It looks like this is related to making the QAppThreadCalls non-blocking in PR #32204. The workspace name is stored in the `_args` in the QAppThreadCall class and it must be being overwritten by the second workspace name before the first `replace_workspace` is called in the sliceviewer presenter. Hence it gets called with the second name twice.
> 
> I'm not sure from that PR why it was made non-blocking.

The blocking call in #32204 was necessary because a `Workspace::readLock` call needed to be introduced in the [sliceviewer](https://github.com/mantidproject/mantid/blob/1c7f3b04c58c14117f8dfd9633fdd0974be46495/qt/python/mantidqt/mantidqt/widgets/sliceviewer/presenters/presenter.py#L392) so that the `LoadInstrument` calls would cause the sliceviewer to wait for anything writing to the workspace to finish before reading it. Removing the `blocking` causes the old `LoadInstrument` in a loop bug to reappear.

I need to figure out why the route through the replace calls is different for this QxQz case vs the LoadInstrument case.

mantid-builder, 2022-09-12:
Fixed by #34337

PR #34337 (2022-09-12): SliceViewer: Fix a bug when workspaces are replaced from an algorithm with more than 1 output
**Description of work.**

Fix a bug with the ordering of arguments when a method call was made from a no-gui to the gui thread.

See the first commit details for a fuller description.

**To test:**

See the original issue and the problem should be fixed.
Examine the unit test to confirm it covers the case.
Try the case in #31367 to confirm the bug has not reemerged.

Fixes #34125

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingA [... cut]
closed 2022-09-12 by mantid-builder