mantidproject/mantid #34126: CaChannel import fails on idaaas

gemmaguest, 2022-06-20:
**Original reporter:** Max at ISIS

**Describe the bug**
The ISIS reflectometry GUI has live data functionality that relies on the CaChannel library. The import currently fails on idaaas reflectometry instances with the default installed CaChannel version.

**To Reproduce**

- Run `import CaChannel` in a script or the python interpreter.
- You get this error:

```
SystemError: ffi_prep_closure(): bad user_data (it seems that the version of the libffi library seen at runtime is different from the 'ffi.h' file seen at compile-time)  File "/home/gg1082848/Desktop/cachannel.py", line 1, in <module>    # import mantid algorithms, numpy and matplotlib  File "/usr/local/lib/python3.6/site-packages/CaChannel/__init__.py", line 3, in <module>    from .CaChannel import CaChannel, CaChannelException  File "/usr/local/lib/python3.6/site-packages/CaChannel/CaChannel.py", line 18, in <module>    from . import ca  File "/usr/local/lib/python3.6/site-packages/CaChannel/ca.py", line 13, in <module>    from caffi.ca import *  File "/usr/local/lib/python3.6/site-packages/caffi/__init__.py", line 3, in <module>    from . import ca  File "/usr/local/lib/python3.6/site-packages/caffi/ca.py", line 146, in <module>    @ffi.callback('void(*)(struct exception_handler_args)')  File "/usr/local/lib64/python3.6/site-packages/cffi/api.py", line 397, in callback_decorator_wrap    error, onerror)
```

**Expected behavior**
The import should work

**Platform/Version (please complete the following information):**
 - OS: idaaas

martyngigg, 2022-09-12:
Changes are currently up on ral-facilities/daaas-ansible-workspace#755.

DanielMurphy22, 2022-09-20:
Can this be closed?

gemmaguest, 2022-09-23:
Max says live data is still not working for him but I'm not sure why. I'll add a new issue for that but link back to this in case it's related.

closed 2022-09-23 by gemmaguest