mantidproject/mantid #34158: Instrument view leak 1

AnthonyLim23, 2022-06-28:
The instrument view has at least 1 memory leak. Identifying any further leaks will be difficult until this one is fixed. 

**Describe the bug**
Open Mantid and make a note of the memory usage. 
Run the script:
```
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np

for i in range(51909,51915):
    LoadRaw(Filename='/archive/NDXWISH/instrument/data/cycle_22_2/WISH000'+str(i)+'.raw', OutputWorkspace=str(i))
    CropWorkspace(InputWorkspace=str(i), OutputWorkspace=str(i), XMin=6000, XMax=99900)
    ConvertUnits(InputWorkspace=str(i), OutputWorkspace=str(i), Target='Wavelength')
```

Clear the ADS, notice that the memory usage has (more or less) returned to normal
Run the script again
Open instrument view on one of the WS's (doesnt matter which)
Close instrument view
Clear the ADS
Notice that the memory usage has gone up (on IDAas I saw an increase from 3.5 to 27)
-->

**Expected behavior**

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: [e.g. Windows, RHEL 7, Ubuntu, macOS]
 - OS Version:
 - Mantid Version [e.g. 6.0.0]

**Additional context**

AnthonyLim23, 2022-06-28:
that is strange! Does that mean the leak is elsewhere?

thomashampson, 2022-06-28:
Two other things I've noticed:
- I don't think you need to run the algorithms, just opening and closing the instrument viewer is enough
- If you delete the ws using the "Delete" button, all is fine. If you use the "Clear" button, that will be problematic.

I can reproduce this with the SXD dataset in the training course data, which saves a bit of time when testing.

thomashampson, 2022-09-01:
For some reason I can only reproduce this in the latest release version (6.4) and not in the nightly.

closed 2025-10-16 by github-actions[bot]