mantidproject/mantid #34183: Indirect spectrum numbers are not right

AnthonyLim23, 2022-07-05:
- Type run 96141
- in the detector grouping section, set the value to "groups" and then 5 (edited) 
- press run
- in the ADS right click and select "show detectors"
- The first row will have 0 for both the index and spectrum number. Version 3.11 will have 0 and 1 respectively.

This change is to revert the behaviour back to be the same as 3.11

https://github.com/mantidproject/mantid/commit/8e51384989d6d14b54353c8e29950ecbd9c37904

AnthonyLim23, 2022-07-05:
Spencer has said "My simple solution to all this would be when creating the ws set spectrumNo = index+1"

SpencerHowells, 2022-07-05:
Should start at 1

sf1919, 2022-07-06:
@AnthonyLim23 I think there are some instructions missing - which GUI is this for?

AnthonyLim23, 2022-07-06:
indirect reduction

sf1919, 2022-07-06:
It's not just the spectrum numbers that are wrong. For some reason we have 6 groups and the final dectector number in the final group is 53 not 52. I think a bit more digging is required to see how it ended up like this.

v3.11 

[image]

v6.4
[image]

PR #34199 (2022-07-14): Correction to Indirect spectrum numbers in Data Reduction when grouped
**Description of work.**

This piece of works reverts the numbering of the spectrum in Indirect Data Reduction when spectra are grouped to how the numbering worked in v3.11. The spectra numbers need to start at 1 rather than 0.

**To test:**
- Open Indirect -> Data Reduction GUI
- Type run 96141
- In the detector grouping section, set the value to "groups" and then set the number of groups to 5
- Press the Run button (you may need to scroll down)
- Right click on the resulting workspace in the ADS and select 'Show Detectors [... cut]
closed 2022-07-14 by thomashampson