mantidproject/mantid #34232: Crash in ApplyAbsorptionCorrections if Corrections file incomplete

sf1919, 2022-07-27:
**Describe the bug**
Whilst testing #34229 a separate crash was found when using a Corrections file with only the ``ass`` contribution (in the example below).

**To Reproduce**
1. In the workbench, make sure that ISIS is the facility and IRIS is the instrument.
2. Open the ```Indirect->Corrections``` interface.
3. In the second tab, `Calculate Paalman Pings`, choose the Sample property by browsing to an IRIS test file, e.g. `ExternalData/Testing/Data/UnitTest/irs26176_graphite002_red.nxs`. Then, modify the Sample Thickness so it's not 0,  e.g. `0.1` cm, and set the Chemical Formula to `C`. 
4. Press `Run` button. A number of workspaces should be created, including the correction one called `irs26176_graphite002_FlatPlate_PP_Corrections_ass`.
5. Save the corrections workspace with a name that the `FileFinder` for Corrections property in the `Apply Absorption Corrections` tab can recognize it (`*_Corrections.nxs` seems to be the way)
2. Pick the Sample property either from a file or already loaded workspace, and select Corrections property by browsing to the freshly saved file.
3. Try to Run. Mantid should abruptly close with no error message.

**Expected behavior**
Instead of closing the user should be provided with an error message that helps them correct the issue.

**Platform/Version (please complete the following information):**
- Mantid 6.4 
- Ubuntu 18.04

**Additional context**
See #34229 for additional background to this issue.

SpencerHowells, 2022-07-28:
@sf1919 @darominski I've had a look at this. The 2 Calculate tabs create the same format Group workspace and the Apply tab checks the number of members in it to decide how many are to be used depending on whther there is a container. The ws look Ok. First run with Calculate PP sample only did indeed crash in Apply. Then I restarted Mantid & tested Calculate MC and that was Ok and also with container. Went back to Calculate PP and that worked with container and then also worked without container. Weird!

Harrietbrown, 2023-01-26:
[image] [Crash in ApplyAbsorptionCorrections if Corrections file incomplete](https://trello.com/c/b9HUiQ8c/27-crash-in-applyabsorptioncorrections-if-corrections-file-incomplete)

robertapplin, 2023-11-07:
I can replicate this, so reopening

SpencerHowells, 2023-11-07:
@robertapplin When you run the Calculate program the Save command creates a Group workspace which contain the individual corrrection terms such as Ass (stated in algorithm instructions). When Applying the corrections it checks the contents of the Group ws for container before applying. This is fully explained in the Apply algorithm. It does not & never has tried to read the individual terms. It's all about checking that it is run correctly.

closed 2023-11-15 by SilkeSchomann