mantidproject/mantid #34308: Add Multiple Scattering For ISIS Powder OSIRIS Scripts

Harrietbrown, 2022-08-24:
The new OSIRIS Scripts contain basic processing of data from the OSIRIS insruments #34257
This Issue is to add Multiple Scattering corrections to the OSIRIS.

Harrietbrown, 2023-01-24:
[image] [Add other corrections for OSIRIS Scripts](https://trello.com/c/Gaa1ZRb2/16-add-other-corrections-for-osiris-scripts)

DannyHindson, 2023-02-13:
It's not clear to me whether the DiscusMultipleScatteringCorrection algorithm is ready to be added into the OSIRIS scripts yet. I don't have any more work planned on it but I've not had any feedback from Sanghamitra yet on the results I sent through ~6 months ago. That feedback might more than just a "Yes" or "No". The DiscusMultipleScatteringCorrection algorithm may need to be run iteratively for example rather than just once.

Note - the diffraction scripts already have an option to enable multiple scattering corrections based on an assumption that the sample scatters isotropically. This switch uses an alternative\more simple algorithm called MayersSampleCorrection (the key point about Discus is that it doesn't require an assumption that the scattering is isotropic). This switch is available to the powder instruments (eg POLARIS) but probably isn't available to OSIRIS yet. So an alternative approach would be to expose this switch to OSIRIS to provide some basic functionality pending us receiving some feedback on the Discus algorithm.

DannyHindson, 2023-03-17:
I agreed with Spencer that the best approach here was to use the DiscusMultipleScatteringAlgorithm but we will assume isotropic scattering to avoid any requirement for an iterative approach. The benefit of the DiscusMultipleScatteringAlgorithm over the MayersSampleCorrection is that it considers multiple scattering from a container. The same extra lines relating to the SampleDetails object referred to in #34311 are relevant here because the multiple scattering correction also cares about the sample material\shape.

To implement this I suggest 
1) add a boolean parameter called "multiple_scattering" to the osiris param mapping file

2) putting the call to Discus in the osiris.py unit in this function. I've included the suggested code from issue https://github.com/mantidproject/mantid/issues/34311 on the assumption you implement the absorption correction first:
```
    def _apply_absorb_corrections(self, run_details, ws_to_correct):
        """
        Generates absorption corrections
        :param ws_to_correct: workspace that needs to be corrected
        :return: A workspace containing the corrections
        """

        container_geometry = self._sample_details.generate_container_geometry()
        container_material = self._sample_details.generate_container_material()
        if container_geometry and container_material:
            mantid.SetSample(
                ws_to_correct,
                Geometry=self._sample_details.generate_sample_geometry(),
                Material=self._sample_details.generate_sample_material(),
                ContainerGeometry=container_geometry,
                ContainerMaterial=container_material,
            )
        else:
            mantid.SetSample(ws_to_correct, Geometry=self._sample_details.generate_sample_geometry(),
                             Material=self._sample_details.generate_sample_material())

        if self._inst_settings.multiple_scattering:
            # Multiple Scattering Correction
            # Assume isotropic S(Q) - provides good approximation to MS intensity even if not true
            X = [1.0]
            Y = [1.0]
            Sofq_isotropic = mantid.CreateWorkspace(DataX=X, DataY=Y, UnitX="MomentumTransfer")
            ws_to_correct = mantid.DiscusMultipleScatteringCorrection(InputWorkspace=ws_to_correct,
                                                                      StructureFactorWorkspace=Sofq_isotropic,
                                                                      OutputWorkspace="MSResults")
            ratio = mantid.mtd['MSResults_Ratio_Single_To_All']
            ws_to_correct *= ratio

        # Absorption correction
        events_per_point = 1000
        ws_to_correct = mantid.MonteCarloAbsorption(InputWorkspace=ws_to_correct,
                                                    EventsPerPoint=events_per_point)
        return ws_to_correct
```

You might have to tune the number of scenarios used in the Discus algorithm to achieve a reasonable run time

MohamedAlmaki, 2023-03-22:
[image] [add multiple scattering for ISIS Powder OSIRIS scripts](https://trello.com/c/StksOiXO/5-add-multiple-scattering-for-isis-powder-osiris-scripts)

SpencerHowells, 2023-03-24:
@MohamedAlmaki This piece of script runs OK. If line          

ws_to_correct=mantid.DiscusMultipleScatteringCorrection(InputWorkspace=ws_to_correct,StructureFactorWorkspace=Sofq_isotropic,OutputWorkspace="MSResults")

is replaced with
mantid.DiscusMultipleScatteringCorrection(InputWorkspace=ws_to_correct,
    StructureFactorWorkspace=Sofq_isotropic,OutputWorkspace="MSResults")

othwewise ws_to_correct is overwritten by MSResults

MohamedAlmaki, 2023-09-14:
Resolved in #35927

closed 2023-09-14 by MohamedAlmaki