mantidproject/mantid #34311: Add absorbtion correction optinon to OSIRIS Scripts

Harrietbrown, 2022-08-25:
The new OSIRIS Scripts contain basic processing of data from the OSIRIS insruments #34257
This Issue is to add the absorbtion correction functionality present in our interfaces to the scripts.

Harrietbrown, 2023-01-24:
[image] [Add other corrections for OSIRIS Scripts](https://trello.com/c/Gaa1ZRb2/16-add-other-corrections-for-osiris-scripts)

DannyHindson, 2023-03-17:
To enable absorption corrections in the standard ISIS powder diffraction scripts:

1) add some lines like the following to the user script to set up the sample details (shape + material):
```
sample_details = SampleDetails(height=4.0, radius=0.2985, center=[0, 0, 0], shape='cylinder')
sample_details.set_material(chemical_formula='Si', number_density=2.0, crystal_density=1.0)
sample_details.set_container(radius=0.3175, chemical_formula='V',number_density=1.0)
#sample_details = SampleDetails(width=4.0, height=4.0, thickness=0.1, center=[0, 0, 0], shape='slab', angle=0.0)
#sample_details.set_material(chemical_formula='Si', number_density=2.0, crystal_density=1.0)
#sample_details.set_container(front_thick=0.3175, back_thick=0.3175, chemical_formula='H2-O',number_density=1.0)
polaris.set_sample_details(sample=sample_details)
```
2) add a setting like this to the config .yaml file:
```
do_absorb_corrections : True
empty_can_subtraction_method : PaalmanPings #Simple
paalman_pings_events_per_point: 10
```

DannyHindson, 2023-03-17:
There are two different absorption corrections algorithms:
1) PaalmanPingsMonteCarloAbsorption - includes empty can subtraction
2) MonteCarloAbsorption - doesn't do empty can subtraction

AnthonyLim23, 2023-03-20:
[image] [add absorption correction](https://trello.com/c/yZ9LzGRn/1-add-absorption-correction)

closed 2023-07-28 by gemmaguest