mantidproject/mantid #34315: Update SimpleShapeDiscusInelastic with container feature

DannyHindson, 2022-08-31:
DiscusMultipleScatteringCorrection now supports container scattering. This task is to update the wrapper algorithm with some extra fields so it too can support containers

SpencerHowells, 2022-08-31:
@DannyHindson Would it be possible to do a container_condition = VisibleWhenProperty...  for the container data?

DannyHindson, 2022-09-01:
> @DannyHindson Would it be possible to do a container_condition = VisibleWhenProperty... for the container data?

I've just created a pull request to add in your container updates to this algorithm. I included some extra features such as the container_condition so the container fields are only visible if you check the "container" field.

There was one thing I spotted and wondered whether I should change. At the moment the sample and container density fields have default values set up (1.0 I think) which means that if you specify an element you can't get Mantid to look up the density in its internal tables. Is it worth removing the default so that the internal densities can be used for elements?

SpencerHowells, 2022-09-01:
@DannyHindson Good idea. Go ahead. I hadn't realised that the tables had densities!

closed 2022-09-08 by martyngigg