mantidproject/mantid #34348: Manual Testing ISIS Indirect interfaces

DanielMurphy22, 2022-09-13:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

Follow the directions at: https://developer.mantidproject.org/Testing/IndirectInelastic/IndirectInelasticAcceptanceTests.html

Harrietbrown, 2022-09-15:
In Elwin the eq2 workspace should have y units of Q but instead it is in units of ln((meV))^{-1}

Harrietbrown, 2022-09-15:
F(Q) fit tab in add workspace with file in the combo box if you change parameter type from width to EISF it crashes mantid.

Harrietbrown, 2022-09-15:
in conv fit tab load all spectra from irs26176_graphite002_red then fit with 2 Lorentzians then save result.
then go to F(q) fit and add workspace. try to add a file loading the nexus you just saved from conv fit. the loading is slow and it does not update the parameter names. but if you change parameter types it will refresh them.

closed 2022-09-20 by DanielMurphy22