mantidproject/mantid #34351: Manual Testing Muon

DanielMurphy22, 2022-09-13:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

See instructions at: https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/index.html
https://developer.mantidproject.org/Testing/MuonInterface/MuonTesting.html

AnthonyLim23, 2022-09-13:
I talked to @martyngigg about this yesterday. Since there have not been any changes for 2 releases and the tests take a very long time, I will just test it for a little bit around my meetings tomorrow.

DanielMurphy22, 2022-09-13:
Okay! Tell me if your team finish early!

AnthonyLim23, 2022-09-14:
Just had a quick play with the 3 main GUI's (ALC, muon analysis, frequency domain analysis) and they seem to work fine. I didnt check everything, but the I tested the main functionality and the things most likely to have been changed.

closed 2022-09-14 by AnthonyLim23