mantidproject/mantid #34354: Manual Testing Workspace Groups

DanielMurphy22, 2022-09-13:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

* Follow the examples at http://docs.mantidproject.org/concepts/WorkspaceGroup.html 
* Ensure MantidWorkbench displays Group Workspaces correctly in the workspace tree

AnthonyLim23, 2022-09-13:
can add a workspace group to a workspace group using a script

RichardWaiteSTFC, 2022-09-13:
> can add a workspace group to a workspace group using a script

Script to reproduce
```
ws1 = CreateSampleWorkspace()
ws2 = CreateSampleWorkspace()

# Create a group workspace and add to the ADS
ws_group = WorkspaceGroup()
mtd.add("group1", ws_group)
ws_group.add("ws1")

ws_group2 = WorkspaceGroup()
mtd.add("group2", ws_group2)
ws_group2.add("ws2")

ws_group.add(ws_group2)  # fails
ws_group.add("group2")  # works - no duplicate
ws_group.addWorkspace(ws_group2)  # works - duplicate of group2
```

closed 2022-09-13 by RichardWaiteSTFC