mantidproject/mantid #34359: Manual Testing Mantid Basics Course

DanielMurphy22, 2022-09-13:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

* Check the Qt-help docs, built into Workbench, against the online documentation 
* Open up the basic course: http://docs.mantidproject.org/tutorials/mantid_basic_course/index.html#mantid-basic-course
* Check that the information and exercises are up to date and work in the Workbench release candidate

DanielMurphy22, 2022-09-13:
-  On Workbench Main Window page it does not mention that the layout boxes can be dragged and reset.

DanielMurphy22, 2022-09-13:
The Load dialog looks for files with certain extensions but this list is smaller on Windows:
[image]
Missing .nxs and .raw most notably!
On Mac:
[image]

DanielMurphy22, 2022-09-13:
On Windows only, highlighting data in the Show Data widget does not highlight the actual data entry boxes, just the columns and rows. You can see the old corect behaviour on the right. [image]

DanielMurphy22, 2022-09-13:
https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/06_formatting_plots.html missing Superplot

robertapplin, 2022-09-13:
Trying to open help documentation for algorithms (e.g. `MinMax`) and interfaces, but it just gives a blank page 
[image]

DanielMurphy22, 2022-09-13:
[image]
Algorithm list inlcudes a lower case rebin v1 as well as the Rebin v1 ( they open the same dialog). There is also a lower case "load" algorithm

robertapplin, 2022-09-13:
Two loads 
[image]

DanielMurphy22, 2022-09-13:
Scripts generated from the algorithm history interface do not have the import of the simpleapi

DanielMurphy22, 2022-09-13:
If you open a script where the first few lines are full, and select yes when prompted to add the mantid API import then the import is added to the end of the file, rather than the start!

DanielMurphy22, 2022-09-14:
We paused just before the connecting data to instruments section

Harrietbrown, 2022-09-14:
When opening the release notes on ubuntu 22 we get n error message. this may be related to the issue wirth help docs we got yesterday.

[image]

Harrietbrown, 2022-09-14:
In sliceViewer load from the training data `CNCS_7860_event` if you select a region of interest and click `Y` or `C` to produce a cut workspace of the readout it fails giving this error:

```
Transpose: EventList: Cannot set Y or E data, these data are generated automatically based on the events
Error in execution of algorithm Transpose:
Transpose: error (see log)
Transpose-v1: Transpose: error (see log)
```

Harrietbrown, 2022-09-14:
from error reporter in testing:

```
Name: Daniel Murphy  Email: [daniel.murphy@stfc.ac.uk](mailto:daniel.murphy@stfc.ac.uk)
       Additional text:
       Opened md_4D from Sliceviewer testing in sliceviewer, selected a region of interest and then clicked the line toolbar button.
       Stack Trace:
       Traceback (most recent call last):
 File "/Applications/MantidWorkbenchUnstable.app/Contents/Resources/lib/python3.8/site-packages/mantidqt/widgets/sliceviewer/views/view.py", line 94, in non_axis_aligned_cuts_clicked
   self.presenter.non_axis_aligned_cut(state)
 File "/Applications/MantidWorkbenchUnstable.app/Contents/Resources/lib/python3.8/site-packages/mantidqt/widgets/sliceviewer/presenters/presenter.py", line 341, in non_axis_aligned_cut
   self._cutviewer_presenter.show_view()
 File "/Applications/MantidWorkbenchUnstable.app/Contents/Resources/lib/python3.8/site-packages/mantidqt/widgets/sliceviewer/cutviewer/presenter.py", line 26, in show_view
   self.reset_view_table()
 File "/Applications/MantidWorkbenchUnstable.app/Contents/Resources/lib/python3.8/site-packages/mantidqt/widgets/sliceviewer/cutviewer/presenter.py", line 37, in reset_view_table
   *self.model.get_default_bin_params(self._sliceview_presenter.get_dimensions(),
 File "/Applications/MantidWorkbenchUnstable.app/Contents/Resources/lib/python3.8/site-packages/mantidqt/widgets/sliceviewer/cutviewer/model.py", line 24, in get_default_bin_params
   vectors = self.proj_matrix[argsort(states), :]
 File "<__array_function__ internals>", line 180, in argsort
 File "/Applications/MantidWorkbenchUnstable.app/Contents/Resources/lib/python3.8/site-packages/numpy/core/fromnumeric.py", line 1120, in argsort
   return _wrapfunc(a, 'argsort', axis=axis, kind=kind, order=order)
 File "/Applications/MantidWorkbenchUnstable.app/Contents/Resources/lib/python3.8/site-packages/numpy/core/fromnumeric.py", line 54, in _wrapfunc
   return _wrapit(obj, method, *args, **kwds)
 File "/Applications/MantidWorkbenchUnstable.app/Contents/Resources/lib/python3.8/site-packages/numpy/core/fromnumeric.py", line 43, in _wrapit
   result = getattr(asarray(obj), method)(*args, **kwds)
TypeError: '<' not supported between instances of 'NoneType' and 'int'
       Using: workbench 6.4.20220912.1832 on macO 12.5.
```

Harrietbrown, 2022-09-14:
loading the md_3D from the last comment you can use the cutting tool, doing so opens a plotting window. this plotting window can be zoomed into by leftclick dragging, but unlike other plots you can not right click drag to zoom out. you need to click home.

The plot also does not update with axes scales untill you click the home button.

Harrietbrown, 2022-09-14:
The axis range for the line cut does not update when you move the line.

Harrietbrown, 2022-09-14:
The print button for plots on mac is not working, there is no error message or responce from the computer

Harrietbrown, 2022-09-14:
trying to use a wireframe plot could potentially have a warning if you try with very large workspaces in the same way we have a warning for if you try and plot 100 lines

Harrietbrown, 2022-09-14:
from our docs this script generates a plot within a plot.:
```
from mantid.simpleapi import *
import matplotlib.pyplot as plt
from matplotlib.ticker import (MultipleLocator, AutoMinorLocator)

#Example data
exp_decay = CreateSampleWorkspace(Function='User Defined',
                                  UserDefinedFunction='\
                                  name=ExpDecay,Lifetime = 20,Height = 200;name=Gaussian,\
                                  PeakCentre=50, Height=10, Sigma=3',
                                  XMax=100, BinWidth=2)

#Create figure and axes
fig, axes = plt.subplots(ncols=2,nrows=1,subplot_kw={'projection': 'mantid'})

# Plot with errorbars on the left set of axes
axes[0].plot(exp_decay, specNum=1, color='red', label='400 K', linewidth=1.0, drawstyle='steps')
axes[0].set_title('Steps and Grids')
axes[0].xaxis.set_minor_locator(AutoMinorLocator())
axes[0].grid(True, which = 'both', axis = 'both') # major/minor, x/y

# Add an inset, use trial and error to find the right location
inset = fig.add_axes([0.77, 0.70, 0.18, 0.18],projection='mantid') #[left, bottom, width, height]
inset.plot(exp_decay, specNum=5, color='blue', label='Log Peak', marker ='.')
plt.yscale('log') #only affects the most recently called axes
inset.set_xlim(40, 60), inset.set_ylim(5, 20)
inset.xaxis.set_minor_locator(AutoMinorLocator()) #inserts 5 minor b/w each major
inset.tick_params(which='minor', width = 0.5, length=4, color='b', direction='in', top='on')
inset.set_title('Inset')

#Plot on the right set of axes
axes[1].errorbar(exp_decay, specNum=1, capsize=2.0, errorevery=5, color='green', label='spec 1', linestyle='--')
axes[1].set_xlabel('Time-of-flight ($\mu s$)', fontsize = 12, fontstyle = 'italic', fontweight = 'bold')
axes[1].set_ylabel('Counts ($\mu s$)$^{-1}$')
axes[1].set_title('Errorbars')

#Use tight layout for subplots and create a legend
plt.tight_layout()
fig.legend(loc='center right')

fig.show()
```

trying to generate a script fromt his plot causes an error warning.

DannyHindson, 2022-09-14:
> In sliceViewer load from the training data `CNCS_7860_event` if you select a region of interest and click `Y` or `C` to produce a cut workspace of the readout it fails giving this error:
>
> ``` [...]

I think there is already an issue for this here:
https://github.com/mantidproject/mantid/issues/33394

DanielMurphy22, 2022-09-14:
We should check the Notebook can be launched on Linux and MacOS directly from the bundled python in the package tomorrow.

DanielMurphy22, 2022-09-15:
As far as I'm aware, all issues have been raised

closed 2022-09-15 by DanielMurphy22