mantidproject/mantid #34361: Manual Testing Project Recovery

DanielMurphy22, 2022-09-13:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

http://developer.mantidproject.org/Testing/ErrorReporter-ProjectRecovery/ProjectRecoveryTesting.html

SilkeSchomann, 2022-09-15:
In step 8. the ordered_recovery.py script should only contain the Load command and this is working fine. However, when clicking on 'Show History' there is not only the Load command but also a few other commands, for instance 'SaveNexus'. Is that ok or do I better raise a GitHub issue for this?

DanielMurphy22, 2022-09-15:
I think this is okay as Show History should show the full history in this case, but oredered_recovery.py should not!

closed 2022-09-15 by DanielMurphy22