mantidproject/mantid #34365: Instrument view crashes when overlaying peaks in Side by Side view

RichardWaiteSTFC, 2022-09-13:
**Original reporter:**Michael Ritchie (SXD student)

**Describe the bug**
Instrument view hard-crashes when overlaying peaks in side-by-side view (at least for SXD data).

**To Reproduce**
(1) Run this script to load a run (from system test data) and make peaks
```
Load(Filename=r'SXD23767.raw', OutputWorkspace='SXD23767')
CreatePeaksWorkspace(InstrumentWorkspace='SXD23767', NumberOfPeaks=0, OutputWorkspace='SingleCrystalPeakTable')
AddPeak(PeaksWorkspace='SingleCrystalPeakTable', RunWorkspace='SXD23767', TOF=3271, DetectorID=32615)
```
(2) Open `SXD23767` in instrument view (right-click on ws in ADS > Show INstrument)
(3) From the drop-down box at the top of the Render tab select `Side by Side`
(4) Drag the `SingleCrystalPeakTable` ws from ADS into the instrument view window

**Expected behavior**
It would be great to support peak overlay in Side by Side mode, from memory I don't think we have ever done this though. In which case I think we should disable the feature properly, or at least catch it and throw a warning.

**Platform/Version (please complete the following information):**
 - OS: Windows - OS Version:
 - Mantid Version: Nightly ~01 Sep 2022

**Additional context**

sf1919, 2022-09-14:
This crash also occurs in v6.4 and v6.3 so in the first instance we'll look to disable the feature for peak overlays.

sf1919, 2022-09-28:
I'd got as far as finding that it is throwing the error on this line but it doesn't seem to be handling that well
https://github.com/mantidproject/mantid/blob/main/qt/widgets/instrumentview/src/PanelsSurface.cpp#L270

mantid-builder, 2022-10-04:
Fixed by #34460

PR #34460 (2022-10-04): Catch error thrown when overlay peaks side-by-side view in instrument view
**Description of work.**
 Small fix to catch error thrown when overlay peaks side-by-side view in instrument view

To-Do
- Add test if possible

**To test:**
(1) Follow instructions on issue - it should now print an error to the log rather than crash

Fixes #34365 

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

- Is the code of an acceptable quality?
- Does the code conform to the [coding standards]( [... cut]
closed 2022-10-04 by mantid-builder