mantidproject/mantid #34366: Possible to add workspace group to another workspace group

RichardWaiteSTFC, 2022-09-13:
**Describe the bug**
Possible to add workspace group to another workspace group - this is not compatible with the project recovery.
Found during manual testing #34354

**To Reproduce**

```
ws1 = CreateSampleWorkspace()
ws2 = CreateSampleWorkspace()

# Create a group workspace and add to the ADS
ws_group = WorkspaceGroup()
mtd.add("group1", ws_group)
ws_group.add("ws1")

ws_group2 = WorkspaceGroup()
mtd.add("group2", ws_group2)
ws_group2.add("ws2")

ws_group.add(ws_group2)  # fails
ws_group.add("group2")  # works - no duplicate
ws_group.addWorkspace(ws_group2)  # works - duplicate of group2
```
**Expected behavior**
Would check that ws is not a group before adding to the group.

**Platform/Version (please complete the following information):**
 - OS: Windows (presumably all)
 - Mantid Version: Nightly 26/08/22 (package for manual testing)

**Additional context**

RichardWaiteSTFC, 2022-09-13:
> ws1 = CreateSampleWorkspace()
> ws2 = CreateSampleWorkspace()
>
> # Create a group workspace and add to the ADS
> ws_group = WorkspaceGroup()
> mtd.add("group1", ws_group)
> ws_group.add("ws1")
>
> ws_group2 = [...]

I can reproduce this in mantid v6.3 - so not a regression. Should we change to release 6.6?

jclarkeSTFC, 2024-10-31:
Some of the tests assume that it's possible to add a group to a group, but should we change the behaviour of `WorkspaceGroup` to not allow this?

Btw the line below fails because `add` takes a string
``` python
ws_group.add(ws_group2)  # fails
```

jclarkeSTFC, 2024-11-06:
We think that no further action is required here, because project recovery would work if the `GroupWorkspaces` algorithm were used, and as far as I know project recovery is not setup to work with all API calls, but it does with algorithms. If this is not the case then we can reopen this.

closed 2024-11-06 by jclarkeSTFC