mantidproject/mantid #34371: Latex content not displaying in algorithm docs

DannyHindson, 2022-09-14:
eg the documentation for DiscusMultipleScatteringCorrection shows several of the following types of error

[image]

thomashampson, 2022-09-16:
I think this is the same as #34118, which should be resolved when we update our Linux builders with new docker images.

martyngigg, 2022-09-26:
I can confirm this works on the latest nightly on macOS. Could someone check Windows?

DannyHindson, 2022-09-26:
I've got the nightly with label 6.4.20220920.0829 installed on Windows (which I think is the latest one) and the latex in the docs is displaying fine now

martyngigg, 2022-09-26:
Excellent. I'm going to close this one as I think it was a machine problem.

closed 2022-09-26 by martyngigg