mantidproject/mantid #34372: Unable to properly replay a history script from ISIS SANS reductions

smk78, 2022-09-14:
**Report to:** @rmdalgliesh , @smk78 

**Describe the bug**
Whilst testing PR https://github.com/mantidproject/mantid/pull/34335 it was observed that the history script failed to fully reload the saved state.

@gemmaguest comments by email:
*I just had a go and I think there are a few issues – one is that it reloads the workspaces with temporary names, so they are hidden by default, and even if you display them you can’t tell what’s what! It also fails to re-open the GUI, which might be related to the same thing, but I’d need to investigate further. And if I manually generate the recovery script, it only seems to include the history for the can workspace and not the sample workspace, which is quite worrying. So unfortunately it is probably quite a bit of work to sort all this out.*
[image]

**Expected behavior**
The history script should fully reproduce the saved state.

**Platform/Version (please complete the following information):**
Release 6.4.0 on IDAaaS & developer environments

gemmaguest, 2023-01-03:
Further to this, if you unroll the history and attempt to re-play it, many of the child algorithms set the output workspace name to "empty_name" (or sometimes "dummy"); we therefore get multiple algorithms overwriting to the same output name, and the name does not get propagated to be used as the input to the next step when replaying the history. 

This was done because interim workspaces are passed around as pointers and don't have a name. However some Mantid algorithms won't run unless you set an output name. Really we should be creating unique `__TMP` names for them but I am not sure if this is unsupported in python, or if some algorithms always force you to set the name anyway (e.g. in some C++ code we use `createTemporaryValue` to create the temporary name where an output name is required, and this function is not exposed to Python).

rbauststfc, 2023-08-23:
We've also found that the history for the Sum Runs tab isn't recording correctly - see issue #35880 where this originally led us to believe there may be a bug. As part of this work, we should also fix the sum runs history.

jhaigh0, 2025-02-24:
Commenting because this came up again, people assigned to this in the future should consider this work https://github.com/mantidproject/mantid/pull/37839 which might help.