mantidproject/mantid #34377: Docs Updates

DanielMurphy22, 2022-09-14:
https://github.com/mantidproject/mantid/issues/34359#issuecomment-1245442421
https://github.com/mantidproject/mantid/issues/34359#issuecomment-1245498654

DanielMurphy22, 2022-09-16:
https://github.com/mantidproject/mantid/issues/34357#issuecomment-1248311915

DanielMurphy22, 2022-10-17:
https://github.com/mantidproject/mantid/issues/34533#issuecomment-1280666930

DanielMurphy22, 2022-10-17:
https://github.com/mantidproject/mantid/issues/34516#issuecomment-1280617951

DanielMurphy22, 2022-10-17:
https://github.com/mantidproject/mantid/issues/34537#issuecomment-1280608573

DanielMurphy22, 2022-10-17:
To install mantidworkbench via conda must add conda-forge to channels. Should add to smoke testing instructions https://github.com/mantidproject/mantid/issues/34528#issuecomment-1280603760

DanielMurphy22, 2022-10-18:
Some of these updates dealt with here: https://github.com/mantidproject/documents/pull/106

The only outstanding docs update for QA is should the Redhat and Ubuntu Smoke testing be merged as they are catered by the same installer now? @martyngigg

martyngigg, 2022-10-19:
> The only outstanding docs update for QA is should the Redhat and Ubuntu Smoke testing be merged as they are catered by the same installer now? @martyngigg

Yes if we could merge those and just label it Linux from now on as it will cover both of those flavours.

PR #34571 (2022-10-25): Docs Updates after release 6.5
**Description of work.**
Add superplot to 06_formatting_plots page
Mention changing layout on gettingstarted/mainwindow basic course 
Add ADS import to 02_pim_sol solution code
Update 03_fit_model_choices to give better fit for EMU data

**To test:**

Build the docs-html target and check that the new links, images and wording are all correct!

Fixes #34377
Fixes #31682

*This does not require release notes* because **minor documentation update**

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/Review [... cut]
closed 2022-10-25 by gemmaguest