mantidproject/mantid #34379: Cutting tool erroneously enabled on 4D workspace after toggling non-ortho view in sliceviewer

RichardWaiteSTFC, 2022-09-15:
Found during testing #34359 (thanks @Harrietbrown and team) and #34363

**Describe the bug**
Cutting tool erroneously enabled on 4D workspace after toggling non-ortho view in sliceviewer (this is probably because the cutting tool is enabled on non-ortho view being turned off without checking again that the workspace supports non-axis cuts).

**To Reproduce**
(1) make the data
```
md_4D = CreateMDWorkspace(Dimensions=4, Extents=[-2,2,-1,1,-1.5,1.5,-0.25,0.25], Names="H,K,L,E", Frames='HKL,HKL,HKL,General Frame',Units='r.l.u.,r.l.u.,r.l.u.,meV')
FakeMDEventData(InputWorkspace=md_4D, UniformParams='5e5') # 4D data
tmp = CreateMDWorkspace(Dimensions=4, Extents=[-0.5,0.5,-1,-0.5,-1.5,-1, -0.25,0], Names="H,K,L,E", Frames='HKL,HKL,HKL,General Frame',Units='r.l.u.,r.l.u.,r.l.u.,meV')
FakeMDEventData(InputWorkspace=tmp, UniformParams='1e5') # 4D data
md_4D += tmp
DeleteWorkspace(tmp)

# Add a non-orthogonal UB
expt_info = CreateSampleWorkspace()
md_4D.addExperimentInfo(expt_info)
SetUB(Workspace='md_4D', c=2, gamma=120)
```

(2) Toggle non-ortho view
(3) Click non-axis cutting tool

```
 File "/Applications/MantidWorkbenchUnstable.app/Contents/Resources/lib/python3.8/site-packages/mantidqt/widgets/sliceviewer/views/view.py", line 94, in non_axis_aligned_cuts_clicked
   self.presenter.non_axis_aligned_cut(state)
 File "/Applications/MantidWorkbenchUnstable.app/Contents/Resources/lib/python3.8/site-packages/mantidqt/widgets/sliceviewer/presenters/presenter.py", line 341, in non_axis_aligned_cut
   self._cutviewer_presenter.show_view()
 File "/Applications/MantidWorkbenchUnstable.app/Contents/Resources/lib/python3.8/site-packages/mantidqt/widgets/sliceviewer/cutviewer/presenter.py", line 26, in show_view
   self.reset_view_table()
 File "/Applications/MantidWorkbenchUnstable.app/Contents/Resources/lib/python3.8/site-packages/mantidqt/widgets/sliceviewer/cutviewer/presenter.py", line 37, in reset_view_table
   *self.model.get_default_bin_params(self._sliceview_presenter.get_dimensions(),
 File "/Applications/MantidWorkbenchUnstable.app/Contents/Resources/lib/python3.8/site-packages/mantidqt/widgets/sliceviewer/cutviewer/model.py", line 24, in get_default_bin_params
   vectors = self.proj_matrix[argsort(states), :]
 File "<__array_function__ internals>", line 180, in argsort
 File "/Applications/MantidWorkbenchUnstable.app/Contents/Resources/lib/python3.8/site-packages/numpy/core/fromnumeric.py", line 1120, in argsort
   return _wrapfunc(a, 'argsort', axis=axis, kind=kind, order=order)
 File "/Applications/MantidWorkbenchUnstable.app/Contents/Resources/lib/python3.8/site-packages/numpy/core/fromnumeric.py", line 54, in _wrapfunc
   return _wrapit(obj, method, *args, **kwds)
 File "/Applications/MantidWorkbenchUnstable.app/Contents/Resources/lib/python3.8/site-packages/numpy/core/fromnumeric.py", line 43, in _wrapit
   result = getattr(asarray(obj), method)(*args, **kwds)
TypeError: '<' not supported between instances of 'NoneType' and 'int'
       Using: workbench 6.4.20220912.1832 on macO 12.5.
```
**Expected behavior**
Cut tool not enabled if workspace doesn't support it

RichardWaiteSTFC, 2022-09-15:
Cutting tool introduced last release so worth fixing for this release

mantid-builder, 2022-09-23:
Fixed by #34400

PR #34400 (2022-09-23): Check workspace supports non axis cuts before enabling on toggle nonortho view off in sliceviewer
**Description of work.**

One-line fix to check workspace supports non axis cuts before enabling on toggle nonortho view off in sliceviewer

**To test:**

(1) Follow instructions in issue

Fixes #34379 

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

- Is the code of an acceptable quality?
- Does the code conform to the [coding standards](http://developer.mantidproject.org/Stand [... cut]
closed 2022-09-23 by mantid-builder