mantidproject/mantid #34380: Help interface does not function on Ubuntu 22

Harrietbrown, 2022-09-15:
During mannual testing it was found that in Ubuntu 22.04 the help interface is empty. We also recived an internal error attempting to open it to the release notes from the welcome page.
[image]

**To Reproduce**
1. either from a clean install welcome page or go to `help->About Mantid Workbench`
2. open the release notes

**Platform/Version (please complete the following information):**
 - OS: Ubuntu
 - OS Version: 22.04

DanielMurphy22, 2022-09-15:
As this is not a problem on Ubuntu 18.04, which isn't yet a supported OS, I think it should not be fixed urgently for this release.

AndreiSavici, 2022-09-15:
I was able to build (and use) the documentation on Ubuntu 22.04. I had to install `python3-sphinxcontrib.qthelp`. Note that I'm not using the conda development environment.

jclarkeSTFC, 2023-11-15:
The help is now working with 6.8 on Ubuntu 22.04

closed 2023-11-15 by jclarkeSTFC