mantidproject/mantid #34382: Sliceviewer Cut viewer 1D plot not updating axes limits after figure interaction

RichardWaiteSTFC, 2022-09-15:
Found during testing #34359 (thanks @DanielMurphy22 for bringing this to my attention)
I'm linking the sliceviewer manual testing issue so we have all the issues in one place #34363

**Describe the bug**
The axes limits are not automatically rescaled after the cut is updated if the 1D plot has been altered (user has panned, zoomed, right-clicked and changed formatting)

Also the right-click option to e.g. change the error bar visibility, axes scale etc. do not automatically update (need to click home button) - this may be related.

**To Reproduce**
(1) Make data (as per manual testing)
```
md_4D = CreateMDWorkspace(Dimensions=4, Extents=[-2,2,-1,1,-1.5,1.5,-0.25,0.25], Names="H,K,L,E", Frames='HKL,HKL,HKL,General Frame',Units='r.l.u.,r.l.u.,r.l.u.,meV')
FakeMDEventData(InputWorkspace=md_4D, UniformParams='5e5') # 4D data
tmp = CreateMDWorkspace(Dimensions=4, Extents=[-0.5,0.5,-1,-0.5,-1.5,-1, -0.25,0], Names="H,K,L,E", Frames='HKL,HKL,HKL,General Frame',Units='r.l.u.,r.l.u.,r.l.u.,meV')
FakeMDEventData(InputWorkspace=tmp, UniformParams='1e5') # 4D data
md_4D += tmp
DeleteWorkspace(tmp)

# make a 3D MDEvent workspace by integrating over all E
md_3D = SliceMD(InputWorkspace='md_4D', AlignedDim0='H,-2,2,100', AlignedDim1='K,-1,1,100', AlignedDim2='L,-1.5,1.5,100')
```
(2) Open `md_3D` in sliceviewer
(3) Open non-axis cut viewer
(4) Pan in the 1D plot
(5) Increase the cut thickess (drag the bottom red marker with white face) to include the high-intensity region at the bottom of the slice
(6) The axes limit will not be updated with larger limits

**Expected behavior**
Plot would rescale axes automatically on update

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: [e.g. Windows, RHEL 7, Ubuntu, macOS]
 - OS Version:
 - Mantid Version [e.g. 6.0.0]

**Additional context**

RichardWaiteSTFC, 2022-09-15:
@martyngigg and @DanielMurphy22 - this makes it difficult to use the cut viewer tool, would it be OK to put it in release 6.5?

martyngigg, 2022-09-15:
@RichardWaiteSTFC Let's make the changes and judge the risk when they are available. Assume for now 6.5 and we can review when the changes are available.

mantid-builder, 2022-09-17:
Fixed by #34395

PR #34395 (2022-09-16): Correctly autoscale axes limits on workspace update non-axis cut viewer in sliceviewer
**Description of work.**

If an artist is updated the ax.autoscale_view() doesn't work without a call to ax.relim() - but that doesn't support containers (such as errorbars) so doesn't get the right limits.

Here I use another autoscale method, ax.autoscale() - but without for it to work on update we have to set ignore_existing_data_limits = True

Note  `ax.autoscale()`  gets set to False if the axes limits are set (e.g. by panning or zooming) - so this needs to be called on every update (not on axes creatio [... cut]
closed 2022-09-17 by mantid-builder