mantidproject/mantid #34386: Imports added to bottom of python script

robertapplin, 2022-09-15:
**Describe the bug**
If you open a script where the first few lines are full, and select yes when prompted to add the mantid API import then the import is added to the end of the file, rather than the start!

**To Reproduce**
1. Open a python script in mantid which has the first few lines full
2. Click yes when prompted
3. The imports are added to the bottom of the file instead of the top

**Expected behavior**
The imports should be added to the top of the file

**Platform/Version (please complete the following information):**
All

DanielMurphy22, 2022-09-16:
This bug has existed for a few releases so dropping from fixing for this release, but it would be great for this one to be fixed!

DanielMurphy22, 2022-09-16:
https://github.com/mantidproject/mantid/blob/main/qt/python/mantidqt/mantidqt/widgets/codeeditor/scriptcompatibility.py

closed 2022-10-03 by DanielMurphy22