mantidproject/mantid #34387: Plot interface generate script to clipboard fails when attempint to generate a plot within a plot

Harrietbrown, 2022-09-15:
In the mantid basics course we provide this script to generate a plot within a plot:
```
from mantid.simpleapi import *
import matplotlib.pyplot as plt
from matplotlib.ticker import (MultipleLocator, AutoMinorLocator)

#Example data
exp_decay = CreateSampleWorkspace(Function='User Defined',
                                  UserDefinedFunction='\
                                  name=ExpDecay,Lifetime = 20,Height = 200;name=Gaussian,\
                                  PeakCentre=50, Height=10, Sigma=3',
                                  XMax=100, BinWidth=2)

#Create figure and axes
fig, axes = plt.subplots(ncols=2,nrows=1,subplot_kw={'projection': 'mantid'})

# Plot with errorbars on the left set of axes
axes[0].plot(exp_decay, specNum=1, color='red', label='400 K', linewidth=1.0, drawstyle='steps')
axes[0].set_title('Steps and Grids')
axes[0].xaxis.set_minor_locator(AutoMinorLocator())
axes[0].grid(True, which = 'both', axis = 'both') # major/minor, x/y

# Add an inset, use trial and error to find the right location
inset = fig.add_axes([0.77, 0.70, 0.18, 0.18],projection='mantid') #[left, bottom, width, height]
inset.plot(exp_decay, specNum=5, color='blue', label='Log Peak', marker ='.')
plt.yscale('log') #only affects the most recently called axes
inset.set_xlim(40, 60), inset.set_ylim(5, 20)
inset.xaxis.set_minor_locator(AutoMinorLocator()) #inserts 5 minor b/w each major
inset.tick_params(which='minor', width = 0.5, length=4, color='b', direction='in', top='on')
inset.set_title('Inset')

#Plot on the right set of axes
axes[1].errorbar(exp_decay, specNum=1, capsize=2.0, errorevery=5, color='green', label='spec 1', linestyle='--')
axes[1].set_xlabel('Time-of-flight ($\mu s$)', fontsize = 12, fontstyle = 'italic', fontweight = 'bold')
axes[1].set_ylabel('Counts ($\mu s$)$^{-1}$')
axes[1].set_title('Errorbars')

#Use tight layout for subplots and create a legend
plt.tight_layout()
fig.legend(loc='center right')

fig.show()
```

From this script if you attempt to generate a script to clipboard to reproduce the plot it causes an error:
```
Traceback (most recent call last):
  File "C:\MantidUnstableInstall\bin\lib\site-packages\workbench\plotting\figuremanager.py", line 459, in generate_plot_script_clipboard
    script = generate_script(self.canvas.figure, exclude_headers=True)
  File "C:\MantidUnstableInstall\bin\lib\site-packages\workbench\plotting\plotscriptgenerator\__init__.py", line 77, in generate_script
    ax_object_var = get_axes_object_variable(ax)
  File "C:\MantidUnstableInstall\bin\lib\site-packages\workbench\plotting\plotscriptgenerator\__init__.py", line 214, in get_axes_object_variable
    num_rows = ax.numRows
AttributeError: 'MantidAxes' object has no attribute 'numRows'
```

DanielMurphy22, 2022-09-16:
I'm undecided as to whether this should be fixed or not. This is a bit of a strange plot that can only be created from a script and the "generate script" button should only support plots generated from the GUI. But then if there was a simple way for us to know beforehand that script generation won't work, then maybe we could disable the "generate script" button.

thomashampson, 2023-06-09:
It doesn't make sense to support anything other than plots that are generated via the GUI. If a plot was created by a script, there's no need to generate a script from the plot. The only thing that might be worth doing is providing a meaningful message to the user.

closed 2023-08-21 by gemmaguest