mantidproject/mantid #34388: Remove MantidNotebook.bat

DanielMurphy22, 2022-09-15:
https://github.com/mantidproject/mantid/issues/34359#issuecomment-1246992328 

The mantidpython and notebook are no longer supported on Linux and MacOS. They should be removed from Windows too. 

Users are expected to create a conda environment with mantid and notebook (or install notebook onto the python inside mantid.)
This docs page should be updated to reflect this: https://docs.mantidproject.org/nightly/concepts/PythonNotebook.html

DanielMurphy22, 2022-09-16:
Considering removing (from this directory https://github.com/mantidproject/mantid/tree/main/buildconfig/CMake/Packaging):

mantidpython.bat.in, (maybe mantidpython.in), AddPythonPath.py.in, deb/scripts and rpm/scripts

I'm also watching the changes here: https://github.com/mantidproject/mantid/pull/34333

DanielMurphy22, 2022-09-23:
Still have "python36-ipython,python36-ipython-notebook" and "jupyter-notebook" in CMakeLists.txt, not sure we need these...

DanielMurphy22, 2022-10-07:
There are a few references to `notebook` in the the Data Processor Widget, which I think are fine because this just calls the algorithms to create a notebook. It was a challenge, but I found this docs page that I think should be in the dev-docs. It is not linked on the docs/interfaces/index page. https://docs.mantidproject.org/nightly/interfaces/framework/DataProcessorWidget.html

@gemmaguest can we have a quick chat about this? Is this DataProcessorWidget currently used in the Reflectometry interface?

DanielMurphy22, 2022-10-07:
Should I remove all things NSIS packaging related??
https://github.com/mantidproject/mantid/blob/main/buildconfig/CMake/NSIS.template.in
https://github.com/mantidproject/mantid/blob/main/buildconfig/CMake/WindowsNSIS.cmake
https://github.com/mantidproject/mantid/blob/main/buildconfig/CMake/WindowsNSISQt5.cmake

gemmaguest, 2022-10-13:
> There are a few references to `notebook` in the the Data Processor Widget, which I think are fine because this just calls the algorithms to create a notebook. It was a challenge, but I found this docs page that I think should be in the dev-docs. It is not linked on the docs/interfaces/index page. [docs.mantidproject.org/nightly/interfaces/framework/DataProcessorWidget.html](https://docs.mantidproject.org/nightly/interfaces/framework/DataProcessorWidget.html)
> 
> @gemmaguest can we have a quick chat about this? Is this DataProcessorWidget currently used in the Reflectometry interface?

No, it is no longer used and I plan to remove the DataProcessorWidget: #33643

closed 2022-11-02 by gemmaguest