mantidproject/mantid #34391: Fix documentation warnings

sf1919, 2022-09-15:
**Describe the bug**
There are more than 20 warnings when building the user documentation. Some of these warnings relate to the ISISPoweder-OSIRIS.rst file which is also not showing in tables of content. It is hoped that by fixing these warnings it may resolve that issue also.

**To Reproduce**
1. Build the documentation and note the warning messages.

**Expected behavior**
There should be no warnings.

**Platform/Version (please complete the following information):**
- release-next branch built on Windows 10 machine on 15/09/22

DanielMurphy22, 2022-09-16:
@AndreiSavici suggested that the OSIRIS_Powder file is "Missing `.. categories:: Techniques` at the end of the file"

mantid-builder, 2022-09-23:
Fixed by #34422

PR #34422 (2022-09-23): Fix documentation warnings
**Description of work.**

- Added `ISISPowder-OSIRIS-v1` to index
- Fixed typos in `ISISPowder-OSIRIS-v1` document cross references
- Removed the following references in `ISISPowder-OSIRIS-v1`, which appear to refer to non-existing sections:
    -  output_directory_osiris_isis-powder-diffraction-ref
    -  vanadium_normalisation_osiris_isis-powder-diffraction-ref
    -  grouping_file_name_osiris_isis-powder-diffraction-ref
    -  input_mode_osiris_isis-powder-diffraction-ref
- Fixed remaining warnings related to documentation, as per #34391 
 
**To test:**
- Build do [... cut]
closed 2022-09-23 by mantid-builder