mantidproject/mantid #34428: Workspace history copy script does not import mantid.simpleapi

Harrietbrown, 2022-09-21:
When generating the history for a workspace and copying it to a script the script does not contain the import of `mantid.simpleapi` that the script will require to run. when importing the script into mantid it is less of an issue because the script editor will ask if you want to add it to the top, but ideally it should include the imports needed to run.

AnthonyLim23, 2022-11-29:
[image] [workspace history script missing import mantid.simpleapi](https://trello.com/c/R8KhhIs2/21-workspace-history-script-missing-import-mantidsimpleapi)

MohamedAlmaki, 2022-12-23:
This is resolved in #34438 
right? @Harrietbrown

Harrietbrown, 2023-01-03:
You are correct.
Fixed by #34438

closed 2023-01-03 by Harrietbrown