mantidproject/mantid #34439: Live data in refl GUI still not working on IDAaaS

gemmaguest, 2022-09-23:
Max has reported that live data is still not working on IDAaaS. I haven't been able to get hold of him to get more details yet; it is working for me. There are a number of things that it could be. See the latest comment on [this issue](https://github.com/ISISNeutronMuon/mantid-isis-support/issues/44) for some of my thoughts. Also see the earlier comment on that issue for other ideas.

@martyngigg @thomashampson I'm away on leave now so could someone contact Max to find out more? If Max is not available it would also be good to contact Becky or another reflectometry scientist to see if it works for them. The first thing to check is that CaChannel is working for them. If it's more a problem with the GUI/algorithm then maybe @rbauststfc could take a look.

thomashampson, 2022-09-26:
I followed up with Max and he said that it is now working.

martyngigg, 2022-09-27:
Closing this as it looks to be working as expected. We can reopen if it comes back.

closed 2022-09-27 by martyngigg