mantidproject/mantid #34440: ScaleRHSWorkspace check box not working

gemmaguest, 2022-09-23:
Max has reported that the ScaleRHSWorkspace check box is not working on the reflectometry GUI - I think he said that it now does nothing. There was a change to the IPFs for a couple of instruments to change the default for this, including INTER which Max is probably using, so that might have caused a problem e.g. maybe the GUI default is inconsistent with the IPF default. See this PR for the IPF change: #https://github.com/mantidproject/mantid/pull/33971

@rbauststfc Could you contact Max and investigate? If Max is not available, Becky can probably help because she asked for the IPF change.

If this is not working then it is a regression so I think is high priority.

rbauststfc, 2022-09-26:
@martyngigg I've had the following reply from Max about this:

> This was about optional arguments for data workspace stitching (i.e. the edit box at the bottom of the settings tab in the NR GUI) and not related to the same option for the transmission runs.
> In the end I have figured it out: it was a change that I was not aware of...previously it was simple a kwarg "ScaleRHSWorkspace" 0 or 1. Now it seems one needs that and "IndexOfReference". This is a bit annoying, but probably fine for the time being.

I'm still trying to figure out how to replicate the issue and why the behaviour has changed (I'm still building up familiarity with the Reflectometry GUI 🙂), but it sounds like Max is not concerned about getting this fixed in time for the release. Would you still like to keep this as a high priority issue?

martyngigg, 2022-09-26:
@rbauststfc Thanks for the update. We can drop the "High Priority" flag.

We don't have a whole lot of critical issues at the moment so would you mind taking a look at what a fix might look like? We could then make an assessment of whether it is too risky to go to `release-next` and put it in `main`?

martyngigg, 2022-10-04:
We've agreed that this is not required for 6.5 so the changes will go to `main`

closed 2022-10-26 by martyngigg