mantidproject/mantid #34458: Loading files in the Muon Analysis interface does not account for case in file name

Harrietbrown, 2022-09-30:
**Original reporter:** Peter Baker peter.baker@stfc.ac.uk

**Describe the bug**
In the Muon Analysis interface when loading data from a file if the case of the instrument name in the file is not correct it prompts the user to change instrument but it can not load the data even after changing the instrument. and example of incorect case would be `musr00035560.nxs` instead of `MUSR00035560.nxs`

**To Reproduce**
1.  obtain a raw nexus file (contact me)
2. open the muon analysis interface.
3. change the instrument in the file name to lower case.
4. attempt to load the data, you should get an error

**Expected behavior**
If the instrument in the file name is lower case it hould still recognise it as the correct instrument.

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: Windows

AnthonyLim23, 2022-10-06:
this line needs changing:
https://github.com/mantidproject/mantid/blob/main/qt/python/mantidqtinterfaces/mantidqtinterfaces/Muon/GUI/Common/load_file_widget/presenter.py#L135

to
```
instrument_from_workspace = self._model.get_instrument_from_latest_run().upper()
```

The issue is that the instrument name is always assumed to be all caps. So we just need to force it early on to allow for lower or mixed case.

martyngigg, 2022-10-06:
Ok. Let's include this in 6.5.

AnthonyLim23, 2022-10-06:
@martyngigg this is a little more involved than I originally thought. The issue stems from the load algorithm automatically changing the filename to be all caps e.g.
```
import os

def create_load_algorithm(filename):
    output_filename = os.path.basename(filename)
    alg = mantid.AlgorithmManager.create("Load")
    alg.initialize()
    alg.setAlwaysStoreInADS(True)
    alg.setProperty("Filename", filename)
    alg.setProperty("OutputWorkspace", output_filename)
    alg.setProperty("DeadTimeTable", output_filename + '_deadtime_table')
    alg.setProperty("DetectorGroupingTable", '__notUsed')
    # Assume if .bin it is a PSI file so return True, else return False
    return (alg, True) if ".bin" in filename else (alg, False)
    
    
alg, psi_data = create_load_algorithm("C:/Users/BTR75544/work/shared/hifi00035972.nxs")
alg.execute()

print(alg.getProperty("Filename").value) # this will return ['C:/Users/BTR75544/work/shared/HIFI00035972.nxs']
```

We can either
1. Work around the issue in the muon code
2. Change how load works so that it doesnt force capitals (this could be a risk if other parts of the code base depend on this behaviour). 

@martyngigg which would you like? I can do option 1 for release next.

martyngigg, 2022-10-06:
Agreed that **2)** is not applicable for `release-next` but would be the proper solution.

Do we know why the upper-case one doesn't work? On Windows the OS literally cannot tell the difference between `C:/Users/BTR75544/work/shared/HIFI00035972.nxs` & `C:/Users/BTR75544/work/shared/hifi00035972.nxs`. The filesystem should be happy with either of these

AnthonyLim23, 2022-10-06:
later on in the code it uses the lower case as a key, so it can never find a match.

martyngigg, 2022-10-06:
> later on in the code it uses the lower case as a key, so it can never find a match.

Ah okay. How risky is the work around?

AnthonyLim23, 2022-10-06:
should be fairly low. Just having a look into it

PR #34489 (2022-10-13): Muon analysis load files without all cap instrument name
**Description of work.**

The muon analysis (and by extension the frequency domain analysis) GUI was unable to load files if the instrument name was not all capitals (i.e. hifi instead of HIFI). The path was taken from the load algorithm, which automatically makes the instrument name all capitals. This caused problems as the dict for storing runs used the non-caps instrument name, so when searching for runs it would never find a match (as the search had all caps). This PR fixes the problem by always returning the capitalisation used in t [... cut]
closed 2022-10-13 by mantid-builder