mantidproject/mantid #34462: Algorithm Property tooltips only display first sentence.

DanielMurphy22, 2022-10-03:
@thomashampson @martyngigg is this bad enough to go into the release?

**Describe the bug**
If a tooltip has a fullstop(maybe other punctuation) in it then that is where the displayed tooltip for an algorithm property ends. 

**To Reproduce**
Open MantidWorkbench and execute the Load algorithm
When the dialog is open, hover over the Property name "OutputWorkspace"
- The full tooltip should be:
```
"The name of the file(s) to read, including the full or relative "
                  "path. (N.B. case sensitive if running on Linux). Multiple runs "
                  "can be loaded and added together, e.g. INST10,11+12,13.ext"
```
- Sadly you can only see the first sentence! Feel free to check another algorithm.

**Expected behavior**
- Full tooltip displayed

**Platform/Version (please complete the following information):**
I've checked MacOS and Windows and even MantidPlot 5.1.1 so this bug has been there a while.

robertapplin, 2022-10-04:
It uses the brief documentation it seems instead of the full documentation. The brief documentation is set here: https://github.com/mantidproject/mantid/blob/15ef1d68a3427cf52dbf91faa2072ca5c5bae02d/Framework/Kernel/src/Property.cpp#L143

robertapplin, 2022-10-04:
The fix would be to change the following line to be
```
widget->setToolTip(QString::fromStdString(prop->documentation()));
```

https://github.com/mantidproject/mantid/blob/1a9b06868651fe4ddff76f664bf55c8ad0fe5f58/qt/widgets/common/src/AlgorithmDialog.cpp#L503

Is this something worth doing for the release @martyngigg , or to be put in main?

martyngigg, 2022-10-04:
> The fix would be to change the following line to be
>
> ```
> widget->setToolTip(QString::fromStdString(prop->documentation()));
> ``` [...]

Thanks for looking at this, Rob. As it is a change and some beta testing has already been done let's put it on `main` to properly align with trying to be more strict about what we merge to `release-next` at this point.

thomashampson, 2022-10-04:
It was changed here, but there's no explanation for it:
https://github.com/mantidproject/mantid/commit/8b3499485b7469655419c8ee965ec5bb2dc6bfc9

martyngigg, 2022-10-04:
Interesting. That reference in the commit is an old one that points to the old trac ticketing system. The link is https://trac.mantidproject.org/mantid/ticket/9006.

It just suggests that some might be too long to show but does at least give an example.

PR #34472 (2022-10-13): Removes the brief documentation functionality from the Property class
**Description of work.**
This PR ensures the full description of a property is used in the algorithm dialog rather than a shorter version which only shows the first sentence.

**To test:**
Open the algorithm dialog of multiple different algorithms
Check that the tooltip of the properties displays the full documentation and not just the first sentence of the documentation

Fixes #34462 

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

## [... cut]
closed 2022-10-13 by martyngigg