mantidproject/mantid #34473: Abins: CRYSTAL Loader fails to load some high-symmetry structures

ajjackson, 2022-10-04:
**Describe the bug**

Abins should be able to read output files from vibrations performed with CRYSTAL17

In some cases CRYSTAL17 prints a coordinates table with a format such as

```
 *******************************************************************************
 ATOMS IN THE ASYMMETRIC UNIT    2 - ATOMS IN THE UNIT CELL:    5
     ATOM             X(ANGSTROM)         Y(ANGSTROM)         Z(ANGSTROM)
 *******************************************************************************
      1 T   6 C     0.000000000000E+00  0.000000000000E+00  0.000000000000E+00
      2 T   1 H     6.362045560100E-01  6.362045560100E-01  6.362045560100E-01
      3 F   1 H    -6.362045560100E-01 -6.362045560100E-01  6.362045560100E-01
      4 F   1 H     6.362045560100E-01 -6.362045560100E-01 -6.362045560100E-01
      5 F   1 H    -6.362045560100E-01  6.362045560100E-01 -6.362045560100E-01
```

Where the T/F column indicate whether an atom is in the minimal basis. (i.e. atoms labelled by F can be reproduced by applying all the symmetry operations to the atoms labelled by T). In other cases this column is missing:

```
 CARTESIAN COORDINATES - PRIMITIVE CELL
 *******************************************************************************
 *      ATOM          X(ANGSTROM)         Y(ANGSTROM)         Z(ANGSTROM)
 *******************************************************************************
    1    12 MG    0.000000000000E+00  0.000000000000E+00  0.000000000000E+00
    2     8 O     2.105000000000E+00  2.105000000000E+00  2.105000000000E+00
    3    12 MG    2.105000000000E+00  0.000000000000E+00  2.105000000000E+00
    4    12 MG    4.210000000000E+00  2.105000000000E+00  2.105000000000E+00
    5    12 MG    0.000000000000E+00  2.105000000000E+00  2.105000000000E+00
    6    12 MG    2.105000000000E+00  4.210000000000E+00  2.105000000000E+00
    7    12 MG    2.105000000000E+00  2.105000000000E+00  4.210000000000E+00
    8    12 MG    4.210000000000E+00  4.210000000000E+00  4.210000000000E+00
    9    12 MG    2.105000000000E+00  2.105000000000E+00  0.000000000000E+00
   10     8 O     0.000000000000E+00  0.000000000000E+00  2.105000000000E+00
   11     8 O     0.000000000000E+00  2.105000000000E+00  0.000000000000E+00
   12     8 O    -2.105000000000E+00  0.000000000000E+00  0.000000000000E+00
   13     8 O     2.105000000000E+00  0.000000000000E+00  0.000000000000E+00
   14     8 O     0.000000000000E+00 -2.105000000000E+00  0.000000000000E+00
   15     8 O     0.000000000000E+00  0.000000000000E+00 -2.105000000000E+00
   16     8 O    -2.105000000000E+00 -2.105000000000E+00 -2.105000000000E+00
```
Unfortunately the parser seems to have been developed against a file with low symmetry, e.g.

```
 *******************************************************************************
 ATOMS IN THE ASYMMETRIC UNIT   14 - ATOMS IN THE UNIT CELL:   14
     ATOM             X(ANGSTROM)         Y(ANGSTROM)         Z(ANGSTROM)
 *******************************************************************************
      1 T   6 C    -4.106740150017E-01  1.819680685508E+00  9.394353218148E-05
      2 T   6 C    -3.728918947516E-01  2.892138552757E-01 -1.105247883929E-04
      3 T   6 C     1.045654346605E+00 -2.892205235180E-01  7.162351916588E-05
      4 T   6 C     1.083446623228E+00 -1.819696748829E+00 -8.128967278884E-05
      5 T   1 H    -1.429142863334E+00  2.208239819511E+00  1.615807676930E-04
      6 T   1 H     9.397497961903E-02  2.221473994697E+00  8.783839731804E-01
      7 T   1 H     9.387346483085E-02  2.221498956490E+00 -8.782329283258E-01
      8 T   1 H    -9.094702585108E-01 -8.925767404261E-02 -8.737573617105E-01
      9 T   1 H    -9.095941128841E-01 -8.942350851330E-02  8.733701250281E-01
     10 T   1 H     1.582098015667E+00  8.930336799071E-02  8.737901789856E-01
     11 T   1 H     1.582317906426E+00  8.934988692903E-02 -8.734474100182E-01
     12 T   1 H     2.101908334411E+00 -2.208306847937E+00 -6.534432344678E-06
     13 T   1 H     5.788121783867E-01 -2.221433142273E+00 -8.784104675169E-01
     14 T   1 H     5.786872953073E-01 -2.221422121290E+00  8.781750914519E-01
```
and so while there is logic to handle the (non)existence of "T", the presence of "F" can break the parser.

**To Reproduce**

Using the Abins Algorithm, load a CRYSTAL17 output where some atoms are equivalent under symmetry operations.

**Expected behavior**

File should be imported succesfully and Abins continue to run.

**Additional context**

Encountered by AJJ while looking into DMOL3 problems. Will get back to this later...

AnthonyLim23, 2022-10-06:
which milestone would you like this marked against?

ajjackson, 2022-10-06:
Not release-next!

ajjackson, 2023-07-18:
Not stale

closed 2023-07-30 by stale[bot]