mantidproject/mantid #34485: Plots with nan spectra produce errors on full list operations

DanielMurphy22, 2022-10-06:
**Original reporter:** ISIS/Gregory Smith

**Describe the bug**
Plotting workspaces where some spectra have all nan Y values can produce list indicies errors on a full list operation when adding/removing errorbars or (sometimes) when clicking the home toolbar button.

This may be a matplotlib bug, but possibly we could work around it, as often plotting data in Mantid is the first time users check their data so won't know if it's all nan values.

**To Reproduce**
Run this script in MantidWorkbench 

```
from mantid.simpleapi import *
import matplotlib.pyplot as plt
from mantid.plots.utility import MantidAxType
from mantid.api import AnalysisDataService as ADS

nan_ws = CreateWorkspace(OutputWorkspace='nan_ws', DataX='1,2,3', DataY='nan,nan,nan', DataE='1,1,1')
valid_ws = CreateWorkspace(OutputWorkspace='valid_ws', DataX='1,2,3', DataY='1,1,1', DataE='1,1,1')

fig, axes = plt.subplots(edgecolor='#ffffff', num='nan_ws-1', subplot_kw={'projection': 'mantid'})
axes.plot(valid_ws, color='#ff7f0e', label='valid_ws: spec 1', wkspIndex=0)
axes.plot(nan_ws, color='#1f77b4', label='nan_ws: spec 1', wkspIndex=0)
axes.tick_params(axis='x', which='major', **{'gridOn': False, 'tick1On': True, 'tick2On': False, 'label1On': True, 'label2On': False, 'size': 6, 'tickdir': 'out', 'width': 1})
axes.tick_params(axis='y', which='major', **{'gridOn': False, 'tick1On': True, 'tick2On': False, 'label1On': True, 'label2On': False, 'size': 6, 'tickdir': 'out', 'width': 1})
axes.set_title('nan_ws, valid_ws')
axes.set_ylim([0.945, 1.055])
legend = axes.legend(fontsize=8.0).set_draggable(True).legend

plt.show()
```
- Open Figure Options (gear toolbar button)
- Navigate to Curves, Errorbars
- untick the Hide error bars checkbox
- Click `Apply to All`
- Get errorreport with this stacktrace:
```
       Traceback (most recent call last):
 File "C:\MantidInstall\bin\lib\site-packages\mantidqt\widgets\plotconfigdialog\curvestabwidget\presenter.py", line 423, in errorbars_apply_to_all
   self.apply_properties()
 File "C:\MantidInstall\bin\lib\site-packages\mantidqt\widgets\plotconfigdialog\curvestabwidget\presenter.py", line 84, in apply_properties
   FigureErrorsManager.update_limits_and_legend(ax, self.legend_props)
 File "C:\MantidInstall\bin\lib\site-packages\workbench\plotting\figureerrorsmanager.py", line 101, in update_limits_and_legend
   ax.relim()
 File "C:\MantidInstall\bin\lib\site-packages\mantid\plots\mantidaxes.py", line 527, in relim
   min_x, max_x, min_y, max_y = datafunctions.get_errorbar_bounds(container)
 File "C:\MantidInstall\bin\lib\site-packages\mantid\plots\datafunctions.py", line 914, in get_errorbar_bounds
   coords = [array[:, 1] for array in y_segments]
 File "C:\MantidInstall\bin\lib\site-packages\mantid\plots\datafunctions.py", line 914, in <listcomp>
   coords = [array[:, 1] for array in y_segments]
IndexError: too many indices for array: array is 1-dimensional, but 2 were indexed
```
**Expected behavior**
Not too sure... maybe all nan spectra should not be plotted in the first place, or else we should handle how they are used in a matplotlib spectra list so they still appear in the legend...

**Platform/Version (please complete the following information):**
Found on:
 - OS: Windows 10
 - Mantid Version 6.4 and beta version for 6.5

DanielMurphy22, 2023-01-04:
I would like to add some info related to nan spectra plotting, found by @smk78. In that case, trying to edit the name of a curve in the Figure Options failed with a similar error message. 
[image]

Possibly if we think it is worth fixing this to handle renaming curves when there are nan spectra, then we may need to raise a separate Github issue as this is not covered by the current PR #34732

sf1919, 2023-01-13:
It looks like this issue has been resolved by another PR. It is no longer possible to recreate the original issue or the editing of the curve name mentioned by @DanielMurphy22 above. I am therefore closing this issue.

closed 2023-01-13 by sf1919