mantidproject/mantid #34486: Incorrect plots as a result of indirect data analysis when loading data as workspaces

MialLewis, 2022-10-06:
**Describe the bug**
When using the `indirect`, `data analysis`, `Elwin` tab to analyse reduce data in the form of a workspace, incorrect `elf` plots are output. This differs to what happens when the data is read in as a file (hence there is a work around)
More specifically:
1. `elf` graph is plotted against `q`, not temperature as expected
2. only 1 line appears on the plot, regardless of the number of groups specified during data reduction.
3. Plot appears to be a bin plot, as opposed to a spectra plot.

**To Reproduce**
1.  Make sure you have access to the ISIS data archive
2. Open the `data reduction` interface via interfaces > indirect > data reduction.
3. Input runs `26173-26174`, enter to load.
4. Under `detector grouping`, set the grouping to `groups`
5. Set the `Number of Groups` to be `3`
6. Click `Run`
7. Two workspaces are generated, save these workspaces using File > Save Project (to be used later)
8. Open the `data analysis` interface via interfaces > indirect > data analysis
9. On the `Elwin` tab, tab over to `workspace` and add to two generated workspaces, selecting `All Spectra`
10. Click Run
11. Right click on the new workspace created that is appended `elf`
**12. Observe that you can only plot as a bin plot, do so.
13. Observe the x axis is `q` - it is expected to be temperature
14. Observe there is only 1 line, not the 3 expected (it's a bin plot)**
15. Clear all mantid workspaces
16. On the `Elwin` tab, tab back over to `File`, loading the previously saved files
17. Repeat steps `12-14` and observe correct behaviour (note that for this dataset the y axis for `elf` is run No, I believe as the data does not have temperature logged)

SpencerHowells, 2022-10-06:
@MialLewis The data you are using do not have temperature values. Hence the plotting by Run no.

AnthonyLim23, 2022-10-06:
https://github.com/mantidproject/mantid/blob/main/qt/scientific_interfaces/Indirect/IndirectDataAnalysisElwinTab.cpp#L250

MialLewis, 2022-10-06:
> @MialLewis The data you are using do not have temperature values. Hence the plotting by Run no.

yes, It was perhaps a bad dataset to choose to illustrate the bug. The issue is that both of the graphs in the image should be the same (if there was a temperature associated with the dataset x axis should be temperature on both)

SpencerHowells, 2022-10-06:
There are a few issues here. These days there will almost always be an environment so the run no option will hardly occur. The environment should not be hard-wired as Temperature it will be read from the Sample log file. Eg I’ve used magnetic field.
The Temp plot should then an axis title of Temperature (K) eg.
Using the File input tab it groups properly when option ticked. When plotting the _elf ws it asks for Spectrum number which will actually be a Q plot for that Temp. Plotting by Temp has to be done manually by selecting bin in the ws. Ideally the gui should have an extra option to plot it by Bin. I usually do a 3D plot which could also be an option. There is a further option to normalise to the lowest temp.
For the input from workspace it looks as if there is no Group option.

AnthonyLim23, 2022-10-14:
group = merge workspaces

SpencerHowells, 2022-10-17:
Running from the Workspace tab the resolution width is not read from the instrument parameters to give the default integration range like the File tab.

PR #35026 (2023-01-30): 34486 fix elwin workspace run
**Description of work.**

Thsi PR fixes a bug in the Elwin interface that cause it to not process spectra added through the workspace input method properly. instead it only ran for a single selected workspace for all spectra instead of only the spectra chosen from multiple workspaces.

**To test:**

Repeat steps to recreate from the initial issue. test that the plots look as expercted.

Fixes #34486
Fixes #34639

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code [... cut]
closed 2023-01-30 by martyngigg