mantidproject/mantid #34488: Refactor Indirect getInstrumentDetail function to make it safer

robertapplin, 2022-10-06:
**Describe the outcome that is desired.**
There is a widely used function in indirect data reduction called `getInstrumentDetail` which throws runtime_errors when validation fails, and these errors are often uncaught. This isn't very safe, and could be dealt with in a more safe manor.

**Describe any solutions you are considering**
I have considered using `std::optional` to return `std::nullopt` when an instrument detail cannot be found. This should then be handled from each location that calls `getInstrumentDetail`. We could then put one `try-except` claus around the validation method and display a message box before returning `std::nullopt`.

Harrietbrown, 2023-01-26:
[image] [Refactor Indirect getInstrumentDetail function to make it safer](https://trello.com/c/wH80agKL/28-refactor-indirect-getinstrumentdetail-function-to-make-it-safer)

AnthonyLim23, 2023-01-30:
can we have some steps to reproduce this?

closed 2023-08-07 by stale[bot]