mantidproject/mantid #34505: Opening help causes mantid workbench to crash

AndreiSavici, 2022-10-13:
**Describe the bug**

Opening help causes mantid workbench to crash on analysis.sns.gov

**To Reproduce**
Open a conda installed version of mantidworkbench. Go to Help, then click on Mantid Help. The help dialog is empty. When you close it, it causes the workbench to crash.

**Expected behavior**

I want to see the help. Even if there is an error, I want to be able to continue my work.

**Screenshots**

```MantidHelpWindow-[Debug] Trying "/usr/local/anaconda/envs/mantid-dev/bin/MantidProject.qhc"
MantidHelpWindow-[Debug] QHelp Collection file /usr/local/anaconda/envs/mantid-dev/bin/MantidProject.qhc not found
MantidHelpWindow-[Debug] Trying "/usr/local/anaconda/envs/mantid-dev/share/doc/MantidProject.qhc"
MantidHelpWindow-[Debug] Using collection file "/usr/local/anaconda/envs/mantid-dev/share/doc/MantidProject.qhc"
MantidHelpWindow-[Debug] Failed to determine help cache file location
MantidHelpWindow-[Debug] Removing help cache file "/SNS/users/3y9/.local/share/data/mantidproject/MantidProject.qhc"
MantidHelpWindow-[Debug] Loading /usr/local/anaconda/envs/mantid-dev/share/doc/MantidProject.qhc
MantidHelpWindow-[Debug] Making local cache copy for saving information at /SNS/users/3y9/.local/share/data/mantidproject/MantidProject.qhc
MantidHelpWindow-[Warning] Failed to copy collection file
MantidHelpWindow-[Debug] Cannot unregister index tables in file /usr/local/anaconda/envs/mantid-dev/share/doc/MantidProject.qhc.
MantidHelpWindow-[Debug] helpengine.setupData() returned 1
MantidHelpWindow-[Debug] open help window for "qthelp://org.mantidproject/doc/index.html"
[VGL] ERROR: in getGLXDrawable--
[VGL]    184: Window has been deleted by window manager
python3.8: src/hb-object-private.hh:154: Type* hb_object_reference(Type*) [with Type = hb_unicode_funcs_t]: Assertion `hb_object_is_valid (obj)' failed.
```

**Platform/Version (please complete the following information):**
 - OS: RHEL7
 - Mantid Version: 6.4.20221010.1624

**Additional context**

gemmaguest, 2022-10-14:
We're trying to establish if this is something broken in the latest release or if it's a problem with the conda install on certain platforms. So far we have:
- RHEL7: The help docs are empty and it crashes with a conda install of nightly
  - Andrei's local non-conda build is fine
  - v6.4 non-conda install is fine
  - Andrei is checking v6.4 conda install...
- IDAaaS (CentOS): works in conda nightly
- Ubuntu 18.04: works in conda nightly
- Ubuntu 22.04: The help docs are empty in a conda install of nightly
  - Some crashes also seen on opening the help window on Rob's machine (an old unstable build; we're not sure how it was installed).
  - The docs are also empty in v6.4 conda install
  - Andrei's non-conda builds on Ubuntu 20.04 and 22.04 work.

AndreiSavici, 2022-10-14:
I have the same issue with conda install of v6.4. Note that this was when connecting through Thinlinc. If I connect with ssh, I just don't see the help pages (no crash)

peterfpeterson, 2022-10-17:
The crashing is only an issue through thinlinc

peterfpeterson, 2022-10-17:
Everything works fine on local machine or ssh. The issue appears to be with the help window itself. I've been able to reproduce it 2 ways (both via thinlinc).

1. start `mantidworkbench`
2. open the help
3. close the help
4. open any algorithm dialog
5. crash

The other way to break things

1. start `mantidworkbench`
4. open any algorithm dialog
3. close the algorithm dialog
2. open the help
6. crash

peterfpeterson, 2022-10-17:
This can also be triggered through ssh. Run `vglrun mantidworkbenchnightly` and repeat the steps for a crash. This is only seen in virtualgl.

peterfpeterson, 2022-10-25:
The issue came from thinlinc. We moved from v2.5.2. to v3.0.2 and the issue went away

closed 2022-10-25 by peterfpeterson