mantidproject/mantid #34520: Windows Workspace/Algorithm Smoke Tests   :man_astronaut: :woman_astronaut: :space_invader:

DanielMurphy22, 2022-10-17:
**Before testing**:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

**Afterwards**:
- Comment below with any issues you came across.
- If no major issues were found, please **close this testing issue**.
- Check the main issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :man_astronaut: :woman_astronaut: :space_invader:

* Load some workspaces, try 5 from different instruments
* Show detectors on these workspaces
* Check Instrument Viewer for these workspaces, follow  documentation for  [Instrument Viewer Widget](https://docs.mantidproject.org/workbench/instrumentviewer.html) and some more detail on the picking tools in the [Basic Course](https://docs.mantidproject.org/tutorials/mantid_basic_course/connecting_data_to_instruments/03_investigating_data.html)
  - [ ] Render tab 
  - [ ] Draw tab 
  - [ ] Pick tab 
  - [ ] Instrument tree tab 
* Test 5 - 10 random algorithms, follow [how to run Algorithms](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/algorithms_workspaces_and_histories/02_algorithms.html#algorithms) and [pick your favourite](https://docs.mantidproject.org/algorithms/)
 - [ ] Help button 
 - [ ] Validation of inputs 
 - [ ] Overwriting workspace 
 - [ ] Run the algorithm 
- [ ] Check right-click options of the workspaces, they are listed [in the documentation](https://www.mantidproject.org/MantidPlot:_The_Workspace_Menu#Save_Nexus).

DanielMurphy22, 2022-10-17:
@DonaldChung-HK  Is this testing task complete with no major issues found?

DonaldChung-HK, 2022-10-17:
Yes, except when it crashes while running the `AddSampleLogMultiple` Algorithm.

closed 2022-10-17 by DonaldChung-HK

DanielMurphy22, 2022-10-17:
Can you give me some steps to recreate the crash with [AddSampleLogMultiple](https://docs.mantidproject.org/nightly/algorithms/AddSampleLogMultiple-v1.html?highlight=addsamplelogmultiple) ?