mantidproject/mantid #34528: MacOS Smoke Tests: Release 6.5

DanielMurphy22, 2022-10-17:
**Before testing**:
- Check this main issue relates to the OS you will test on.
- Tick the box beside a testing issue listed below to mark *you are testing this issue*
- Enter the testing task issue and assign yourself as for a normal Github issue.

**Afterwards**:
- Comment on the testing issue with any issues you came across.
- If no major issues were found, please **close the testing issue**.

If you have any questions please contact the creator of this issue.

### Tick below before testing the related issue

- [ ] #34522

- [ ] #34523

- [ ] #34524

- [ ] #34525

- [ ] #34526

- [ ] #34527

perenon, 2022-10-17:
Hey, we will perform testing along with @eurydice76

perenon, 2022-10-17:
Hi @DanielMurphy22 , struggling finding macOS installation procedure of the release candidate
Can you help ?

perenon, 2022-10-17:
Should we add conda-forge to channels ?

DanielMurphy22, 2022-10-17:
Try: `conda install -c "mantid/label/v6.5.0-rc1" mantidworkbench`

perenon, 2022-10-17:
Still not. FYI we have a M1 chip

martyngigg, 2022-10-17:
We're currently compiling for Intel chips so I expect using an Intel version of Conda would find it.

perenon, 2022-10-17:
Working with Intel conda. 
Must add conda-forge to channels.
Must also specifying python3.8 when creating environment

martyngigg, 2022-10-17:
> Working with Intel conda. Must add conda-forge to channels. Must also specifying python3.8 when creating environment

Oh interesting. I wonder why arm requires you to specify the Python version 🤔

perenon, 2022-10-17:
Correction, you do not need to specify the use of python3.8.
Installing in a new environment (different from the base one) works

perenon, 2022-10-17:
Hey (sorry to add something outside the normal delay of smoke testing, I can move it somewhere else if you want),

On our macOS machine, we cannot access the doc (by clicking ? button on the bottom left of algorithm window)
The window can open but the doc itself does not appear. Do you have the same ?

DanielMurphy22, 2022-10-17:
I have the same on my Conda installed version on MacOS. all the help documentation is blank.

perenon, 2022-10-17:
(once again sorry for the delay, but that was not mentioned in the issues I worked on)