mantidproject/mantid #34536: Ubuntu Clean/Dirty Install Smoke Tests  :soap: :hankey:

DanielMurphy22, 2022-10-17:
* Follow the instructions [here](https://download.mantidproject.org/ubuntu.html) to install the .deb for Ubuntu

**Before testing**:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

**Afterwards**:
- Comment below with any issues you came across.
- If no major issues were found, please **close this testing issue**.
- Check the main issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :soap: :hankey:

### Dirty install

* Make sure that you have several versions of Mantid installed
 * Last release
 * A nightly
 * If possible an old release
* Install the new release candidate of Mantid
- [ ] Check that Mantid boots up correctly

### Clean install

* Remove all existing Mantid versions and associated files

**On Windows**:
 * Uninstall the program
 * Clear shortcuts from desktop
 * Remove the mantid  files in %APPDATA%

**On macOS** :
 * Remove the application
 * Remove the `~/.mantid directory`
 * Remove (or rename) `~/Library/Preferences/org.python.python.Python.plist`  and `~/Library/Preferences/org.python.python.plist` 

**On Linux** :
 * Remove the package: `/opt/Mantid`
 * Remove `~/.config/Mantid`
 * Remove `~/.mantid/`

Re-install the new release candidate of Mantid

- [ ] Check that Mantid boots up correctly

gui-co, 2022-10-17:
Boot if fine in both cases but I have an issue with the font in the `.tar.xz` installation. Here is a screenshot:
* with conda installation:
[image]
* with `.tar.xz` installation:
[image]

DanielMurphy22, 2022-10-17:
This is seen on all Linux and was the case for the last release as well. Perhaps we could choose a nicer default font but this font in the tar is to cater for all Linux OS: https://github.com/mantidproject/mantid/issues/33875

gui-co, 2022-10-17:
Thanks for the answer.