mantidproject/mantid #34539: Ubuntu Plotting Smoke Tests  :chart_with_upwards_trend: :bar_chart:

DanielMurphy22, 2022-10-17:
* Follow the instructions [here](https://download.mantidproject.org/ubuntu.html) to install the .deb for Ubuntu

**Before testing**:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

**Afterwards**:
- Comment below with any issues you came across.
- If no major issues were found, please **close this testing issue**.
- Check the main issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :chart_with_upwards_trend: :bar_chart:

* 1D plotting:[instructions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/03_displaying_1D_data.html#displaying-1d-data)
 - [ ] Simple plot 
 - [ ] Another way to plot 
 - [ ] Adding curves to existing plots 
 - [ ] Also, test out [waterfall](https://docs.mantidproject.org/nightly/plotting/WaterfallPlotsHelp.html#waterfall-plots) and [tiled]( https://docs.mantidproject.org/nightly/plotting/1DPlotsHelp.html#tiled-plots)
 - [ ] Check Toolbar buttons

* 2D plotting: [instructions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/04_displaying_2D_data.html#displaying-2d-data)
 - [ ] Plot all spectra 
 - [ ] Change colour map 
 - [ ] [Contour plot](https://docs.mantidproject.org/nightly/plotting/ColorfillPlotsHelp.html#contour-plots) (under 3D menu) 
 - [ ] Check Toolbar buttons

* [3D plotting](https://docs.mantidproject.org/nightly/plotting/3DPlotsHelp.html):
 - Load some data eg `LOQ7041` from the ISIS sample data
 - [ ] 3D surface
 - [ ] 3D wire frame  
 - [ ] Check Toolbar buttons

 ## Sliceviewer
 - [ ] Overly long instructions (don't spend  long!) and data [here](https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html). In particular try editing the data in a workspace while it is open in Sliceviewer!

darominski, 2022-10-17:
There is an issue in the help for Sliceviewer, it seems like misformatted math in Latex:
[image] .

Otherwise, everything looks in order.

darominski, 2022-10-17:
This seems to be a wide-spread issue with Latex at the moment, not specific to Sliceviewer, so I am closing this issue.

closed 2022-10-17 by darominski