mantidproject/mantid #34540: Ubuntu Python Smoke Tests  :snake:

DanielMurphy22, 2022-10-17:
* Follow the instructions [here](https://download.mantidproject.org/ubuntu.html) to install the .deb for Ubuntu

**Before testing**:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

**Afterwards**:
- Comment below with any issues you came across.
- If no major issues were found, please **close this testing issue**.
- Check the main issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :snake:

* Test that the Python scripting window works, [directions here](https://docs.mantidproject.org/nightly/workbench/scriptwindow.html)
 - [ ] Editor options 
 - [ ] Execution options 
 - [ ] Script output  
- [ ] Perform some workspace algebra 
- [ ] Test numpy functionality 
- [ ] Use the scripting window to run some scripts 
- [ ] Run through some examples from [the documentation](https://docs.mantidproject.org/nightly/tutorials/python_in_mantid/index.html), 3 or 4 examples from the Solutions is enough

gui-co, 2022-10-17:
The "show in explorer" function does not seem to work. On an unsaved script, a log message says that the script does not correspond to a file but on a properly saved script, nothing happens.

To reproduce:
* open or create a script in mantid (the script has to exist in the file system)
* right click on the script tab and click on "show in explorer"

[image]

I tried on the previous release (6.4) and the function does not work either.

DanielMurphy22, 2022-10-17:
Interesting, I didn't know about this feature! We can fix this after the release!

gui-co, 2022-10-17:
The rest is working as expected.

DanielMurphy22, 2022-10-17:
Close the issue when you're happy!