mantidproject/mantid #34542: Ubuntu Smoke Tests: Release 6.5

DanielMurphy22, 2022-10-17:
* Follow the instructions [here](https://download.mantidproject.org/ubuntu.html) to install the .deb for Ubuntu

**Before testing**:
- Check this main issue relates to the OS you will test on.
- Tick the box beside a testing issue listed below to mark *you are testing this issue*
- Enter the testing task issue and assign yourself as for a normal Github issue.

**Afterwards**:
- Comment on the testing issue with any issues you came across.
- If no major issues were found, please **close the testing issue**.

If you have any questions please contact the creator of this issue.

### Tick below before testing the related issue

- [ ] #34536

- [ ] #34537

- [ ] #34538

- [ ] #34539

- [ ] #34540

- [ ] #34541

darominski, 2022-10-17:
I am afraid that the installation recommendations from the sub-issues are no longer valid, linked with the removal of `.deb` package. How can we get a package, is it already available on conda?

DanielMurphy22, 2022-10-17:
All tests complete I'll make a note to update the instructions for testing as above.

closed 2022-10-17 by DanielMurphy22