mantidproject/mantid #34544: Instrument viewer crashes with DEVA workpace with no facility and IDF

gemmaguest, 2022-10-17:
**Describe the bug**
When loading very old data that does not have a facility and IDF associated with it, you can end up with a workspace that will crash Mantid if you attempt to view it in instrument viewer.

Low priority because users are unlikely to load this old data, but the hard crash should be resolved.

**To Reproduce**

- Load DEVA01360 from the ISIS sample data.
- Show instrument on one of the workspaces. Mantid crashes

**Expected behavior**
It's probably useful to still be able to view very old data like this, so I think we would like to still be able to load the data, but just disable instrument viewer in this case. However if this turns out to be a large amount of work we may want to reconsider whether we support loading data like this.

**Platform/Version (please complete the following information):**
 - OS: Ubuntu, probably all

**Additional context**
We think that this is due to the workspace not having an instrument, so it is hopefully a simple check. However, using `CreateWorkspace` I think creates a workspace with no instrument, and `Show Instrument` is greyed out in this case, so I'm not sure if there is something else going on too. Hopefully catching the crash in the debugger will give more information.
```
ws=CreateWorkspace([1,2,3], [4,5,6])
```

For info, this is an old ISIS instrument that is decommissioned: https://www.isis.stfc.ac.uk/Pages/deva.aspx

jhaigh0, 2023-12-07:
I think this may be the root of the problem
https://github.com/mantidproject/mantid/blob/main/Framework/DataHandling/src/LoadMuonNexus.cpp#L164-L171

sf1919, 2024-01-22:
I'm going to move this one to v6.10 for the moment. Will this potentially be fixed as part of the Nexus epic @robertapplin ?

PR #37665 (2024-07-17): Fix for crash when opening instrument view with old Muon data
### Description of work

https://github.com/mantidproject/mantid/pull/16406 made it possible to load certain old data. As part of this, an empty instrument was added to the workspace by LoadEventMuon which caused a problem when opening the instrument viewer.

This PR adds a check when constructing the instrument renderer to check for an empty instrument.

Fixes #34544

### To test:

- Load `DEVA01360.nxs` from the unit test data (you may have to build the `StandardTestData` target if not already built)
- Attempt to open one of the w [... cut]
closed 2024-07-17 by SilkeSchomann