mantidproject/mantid #34545: Mean algorithm doesn't check input parameters

thomashampson, 2022-10-17:
**Describe the bug**
When running the Mean algorithm from a dialog, if you enter information incorrectly in the input workspaces box it will crash Mantid.

**To Reproduce**
1. No need to load any data
2. Open the Mean algorithm dialog from the algorithm widget
3. Enter something silly in the Workspaces edit box (anything that isn't a currently loaded workspace will do)
4. Enter something in the OutputWorkspace box
Mantid will crash

**Expected behavior**
Input parameters should be checked before attempting to run the algorithm

**Platform/Version (please complete the following information):**
 - OS: found on IDAaaS (not tested yet on Windows)
 - OS Version:
 - Mantid Version: 6.5.0-rc, 6.4.0, not tested on anything earlier

sf1919, 2022-10-17:
Related to #34476 ?

jhaigh0, 2022-10-17:
> Related to #34476 ?

Yes, the workspace names aren't validated in `Mean.validateInputs()` before calling `mtd[name]`. `MeanTest.py` shows the current expect behaviour is a run-time error.

PR #34556 (2022-10-20): Mean algorithm input workspace validation
**Description of work.**

- Added a check to see if the input workspaces exist before trying to access them. If any don't exist then a relevant issue is returned.
- Updated the tests to check exceptions more explictly
- The updated tests uncovered a problem with `_are_workspaces_compatible()` so that has been changed too

**To test:**

- Follow steps to reproduce #34545 and check that an error reporting an invalid workspace name is shown rather than a crash
- Follow steps to reproduce #34476 and check that the Mean algorithm can be opened with the corr [... cut]
closed 2022-10-20 by martyngigg