mantidproject/mantid #34560: Provide more intuitive side by side instrument view for SXD

DannyHindson, 2022-10-18:
The SXD users would like a 2D view of the instrument in the Mantid instrument viewer that is able to represent their 11 banks without any distortion from a projection onto either a cylinder or spherical surface. The requirement is something like the existing side by side view but without the unusual orientation of some of the banks:

[image]

RichardWaiteSTFC, 2022-10-18:
There is also the 3D view in SXD2001 (3D spherical projection) 
[image]

robertapplin, 2022-11-18:
It might be worth running changes past Liliana De Campo to see if this work will help her too 

https://github.com/mantidproject/mantid/issues/34760

DannyHindson, 2022-12-13:
SXD detector layout discussed with Silvia:

[image]

mantid-builder, 2023-02-13:
Fixed by #35090

mantid-builder, 2023-02-14:
Fixed by #35090

closed 2023-02-14 by mantid-builder