mantidproject/mantid #34561: HKL in sliceviewer of cursor info widget

RichardWaiteSTFC, 2022-10-18:
**Is your feature request related to a problem? Please describe.**
WISH scientists want to report HKL of cursor position - in particular MD workspaces with basis vectors e.g. 110^001

**Describe the solution you'd like**
Report HKL in cursor info widget

**Describe alternatives you've considered**
Can add a peak to a table but this is a very clunky workaround

DannyHindson, 2022-10-24:
Just checking I've understood this. So the requirement is for the cursor to give the HKL values for MD workspaces such as this one where the x and y columns in the table in the top right corner do not correspond to H,K or H,L or K,L?

```
ws = CreateMDWorkspace(Dimensions='3', Extents='-3,3,-3,3,-3,3',
                       Names='H,K,L', Units='r.l.u.,r.l.u.,r.l.u.',
                       Frames='HKL,HKL,HKL',
                       SplitInto='2', SplitThreshold='10')

# add fake Bragg peaks for primitive lattice in data
for h in range(-3,4):
    for k in range(-3,4):
        for l in range(-3,4):
            hkl = ",".join([str(x) for x in [h,k,l]])
            FakeMDEventData(ws, PeakParams='1e+02,' + hkl + ',0.02', RandomSeed='3873875')

BinMD(InputWorkspace=ws, AxisAligned=False,
    BasisVector0='[00L],U,0,0,1',
    BasisVector1='[HH0],U,1,1,0',
    BasisVector2='[-HH0],U,-1,1,0',
    OutputExtents='-4,4,-4,4,-0.25,0.25',
    OutputBins='101,101,1', OutputWorkspace='BinMD_out', NormalizeBasisVectors=False)
```

[image]

If we have a workspace that does have basis vectors that are H,K (for example) is it OK to still show the cursor readout for HKL? Same data will also be shown in the x, y columns I think

RichardWaiteSTFC, 2022-10-24:
> Just checking I've understood this. So the requirement is for the cursor to give the HKL values for MD workspaces such as this one where the x and y columns in the table in the top right corner do not correspond to H,K or H,L or K,L?

Yep, so in this case if the cursor in the image was at (X,Y) = (1,2) and the slicepoint at 3 along [-1,1,0] then the HKL would be [1,1,2] + [-3,3,0] = [-2, 4, 2].

> If we have a workspace that does have basis vectors that are H,K (for example) is it OK to still show the cursor readout for HKL? Same data will also be shown in the x, y columns I think

I think that's fine to show it for all HKL workspaces - it won't show anything particularly interesting for XYZ=HKL but at least the cursor info widget would be consistent and we wouldn't need to worry about the basis vectors of the workspace changing while the sliceviewer is open -does that make sense?

DannyHindson, 2022-10-24:
Yes that's all clear thanks

RichardWaiteSTFC, 2022-10-25:
> Just double checking the example you gave. Should the HKL be (-1,5,1)? 1*(0,0,1) + 2*(1,1,0) + 3*(-1,1,0) = (-1,5,1)

Yes sorry I got X and Y axis mixed up - a very unhelpful typo!

closed 2022-11-08 by gemmaguest