mantidproject/mantid #34584: Issues with recent Sphinx versions (above 5.1.1)

thomashampson, 2022-10-20:
**Describe the bug**
- Saving/referencing math images files in the wrong place
- Website search

**To Reproduce**
Remove the 5.1.1 sphinx pin and build the docs package (5.3.2 is definitely not working).

**Expected behavior**
- Latex math should render correctly in our documentaion
- Search should work on our websites

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: [e.g. Windows, RHEL 7, Ubuntu, macOS]
 - OS Version:
 - Mantid Version [e.g. 6.0.0]

**Additional context**

sf1919, 2022-11-07:
Waiting for v6 of Sphinx to be released before trying to remove the pin - https://www.sphinx-doc.org/en/master/changes.html

sf1919, 2022-11-28:
Looks like the earliest v6 will be released is 01/12/22 - https://github.com/sphinx-doc/sphinx/issues/10920

sf1919, 2023-02-16:
Requires pygments to be `>=2.12`, but this can only be achieved if cppcheck is v2.8 or higher.

sf1919, 2023-07-14:
Blocked by https://github.com/mantidproject/mantid/issues/33510

PR #35928 (2023-10-27): Unpin sphinx and pygments
Unpinning sphinx required that pygments also be unpinned, which in turn required cppcheck>=2.8, which is now the case. At the time of writing sphinx is version 7.2.*.

**Description of work**
Unpinned sphinx and pygments. To get the new version to work I had to add the definition of LC_ALL to the build and package scripts. This prevents an error about locales not being available, see #35504 for a similar problem. I also had to edit the python code that generates the rst for the "image not found" section of the doc pages. It looked like new lines were being parsed in a [... cut]
closed 2023-10-27 by SilkeSchomann